sparkmicro/Ki-nTree #4: Demo

WhiteChairFromIkea, 2020-08-29:
Create a demo platform (web server / virtual machine image) for easier start. Just leaving this as low priority to do here 🙂

eeintech, 2020-08-29:
@WhiteChairFromIkea I agree, the entry level for installing the application is quite high. I was looking into [Beeware](https://beeware.org/) software packaging suites to create an installer, is that what you had in mind? Or did you mean more like a [Docker](https://docs.docker.com/) image/container?

WhiteChairFromIkea, 2020-08-31:
I meant you could preconfigure a small linux installation inside Oracle "Virtual box" and provide *.VHD (virtual disk image) for users to download. Users of interest should then install Virtual box and use VHD you provided in order to try things. Your VHD would follow latest changes, because you should reinstall latest version inside your VHD (also - this would be good for you as a troubleshooter).

I think there are no licensing issues - [Oracle "Virtual box"](https://www.virtualbox.org/wiki/Licensing_FAQ) is under GPL2, small linux distro like [Puppy Linux](http://puppylinux.com/index.html) shrinks to under 100Mb (as far as I remember). You could share VHD in your google drive. Virtual box runs on any platform. Installing Virtual box on user OS makes similar impact as would installing all the dependencies on windows (bash, python..).

eeintech, 2020-08-31:
It might be the first time I hear about Oracle "Virtual box". While I understand how it would all come together, do you have an example application which is packaged in such a way? I would like to get a "feel" for the end user.

eeintech, 2020-09-10:
@WhiteChairFromIkea I started playing with [Gitpod](https://www.gitpod.io/), I can get Ki-nTree running and interact with it, however it rendered useless by the fact that KiCad isn't installed on the system it is running on, it should be ran on the same system. InvenTree could work because the API can connect to any given URL but still I don't think there is any benefit in this "demo" version versus spending the time to set it up on your own system. Does it makes sense to you?

WhiteChairFromIkea, 2020-09-10:
Wow, you gave very precise instructions (especially _port 6080 "Detecting..._"), and I can see VM with Ki-nTree GUI, but don't know what to do next :D, but I am willing to know :)
Demo experience should be like [partkeepr]( https://demo.partkeepr.org/) or [partsbox](https://partsbox.com/demo/parts), but I don't have a knowledge about how to do that, sorry. I also don't have an examples for Oracle VM as demos, this was just an idea, and is similar to what you have done already, instead yours Gitpod is true online version, which is cool.

eeintech, 2020-09-10:
Ki-nTree is focused towards [InvenTree](https://github.com/inventree/inventree), I am not really planning on supporting any other part inventory systems, unless someone would be willing to put time into it.

Do you already have an InvenTree instance running locally or on a server?
Do you use KiCad or are you only interested in the inventory system side-of-things?

WhiteChairFromIkea, 2020-09-10:
> Ki-nTree is focused towards [InvenTree](https://github.com/inventree/inventree), I am not really planning on supporting any other part inventory systems, unless someone would be willing to put time into it.
This issue is about creating user friendly demo, partkeepr and partsbox from previous post were just an examples of how online demos should look like;

> Do you already have an InvenTree instance running locally or on a server?
No, installation on Windows was unsuccessful / too complex, so I created this issue about creating some kind of demo platform for new users (like me);

> Do you use KiCad or are you only interested in the inventory system side-of-things?
I am a KiCad user, currently tied to PartsBox. PartsBox free plans are not sufficient when running two or more PCB assembly batches at a time, so I wanted to try InvenTree / Ki-nTree and faced difficulties as a starter.

eeintech, 2020-09-10:
@WhiteChairFromIkea Sorry for being so interrogative and thank you for answering thoroughly, I am was trying to get a sense of your target setup and application.

I understand you are looking for an existing InvenTree demo setup to play with so that you can use Ki-nTree to add parts to, I had assumed you only needed the latter :grimacing: 
InvenTree and @SchrodingersGat (the creator) do not provide demo instance of InvenTree. I think it would be a good idea to have but I can see many problems to hosting a write-enable instance accessible by any user off the internet, it could get really messy. A read-only version like PartsBox or PartKeepr provide is more likely, but in that case that wouldn't solve anything on your side...

Do you have a Rasperry Pi or old laptop you can convert to Linux and run InvenTree on?

eeintech, 2020-09-10:
> Edit: I realized just now... I have failed to setup inventree server, far away to from trying Ki-nTree. It seems I have created an issue in wrong github project :)

That's totally fine! Sorry I couldn't help you more, happy to brainstorm here or there :smile:

eeintech, 2020-09-16:
@WhiteChairFromIkea What do you think, can

closed 2020-09-16 by eeintech