sparkmicro/Ki-nTree #15: Fake user agent error

pawcio0928, 2021-01-26:
Hello I tried to launch Ki-nTree on my MacBook Pro. After done the installation step I tried to test the connection with Digi-Key. Unfortunately it doesn't work and I get this errors. I received in my browser Auth code. 
Second Warning appear after I clicked Allow in my browse. 

```
2021-01-26 11:49:05,373 - digikey.oauth.oauth2 -  WARNING: Oauth2 token storage does not exist or malformed, creating new.
WARNING:digikey.oauth.oauth2:Oauth2 token storage does not exist or malformed, creating new.
WARNING:fake_useragent:Error occurred during loading data. Trying to use cache server https://fake-useragent.herokuapp.com/browsers/0.1.11
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/urllib/request.py", line 1319, in do_open
    encode_chunked=req.has_header('Transfer-encoding'))
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/http/client.py", line 1252, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/http/client.py", line 1298, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/http/client.py", line 1247, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/http/client.py", line 1026, in _send_output
    self.send(msg)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/http/client.py", line 966, in send
    self.connect()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/http/client.py", line 1422, in connect
    server_hostname=server_hostname)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ssl.py", line 423, in wrap_socket
    session=session
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ssl.py", line 870, in _create
    self.do_handshake()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ssl.py", line 1139, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:1076)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/pawel/magazyn/Ki-nTree/env-kintree/lib/python3.7/site-packages/fake_useragent/utils.py", line 67, in get
    context=context,
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/urllib/request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/urllib/request.py", line 525, in open
    response = self._open(req, data)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/urllib/request.py", line 543, in _open
    '_open', req)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/urllib/request.py", line 503, in _call_chain
    result = func(*args)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/urllib/request.py", line 1362, in https_open
    context=self._context, check_hostname=self._check_hostname)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/urllib/request.py", line 1321, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:1076)>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/pawel/magazyn/Ki-nTree/env-kintree/lib/python3.7/site-packages/fake_useragent/utils.py", line 154, in load
    for item in get_browsers(verify_ssl=verify_ssl):
  File "/Users/pawel/magazyn/Ki-nTree/env-kintree/lib/python3.7/site-packages/fake_useragent/utils.py", line 97, in get_browsers
    html = get(settings.BROWSERS_STATS_PAGE, verify_ssl=verify_ssl)
  File "/Users/pawel/magazyn/Ki-nTree/env-kintree/lib/python3.7/site-packages/fake_useragent/utils.py", line 84, in get
    raise FakeUserAgentError('Maximum amount of retries reached')
fake_useragent.errors.FakeUserAgentError: Maximum amount of retries reached
```

I've also problem with executable version. It's only gave me this errors.

```
[INFO]  Warning: Failed to load Digi-Key configuration
[INFO]  Warning: Failed to load InvenTree configuration
[INFO]  Warning: Failed to load KiCad configuration
[ERROR] File /Users/pawel/magazyn/Ki-nTree/dist/config/user_files/kicad.yaml does not exists!
[INFO]  Warning: Failed to load Digi-Key configuration
[INFO]  Warning: Failed to load InvenTree configuration
[INFO]  Warning: Failed to load KiCad configuration

[ERROR] Some Ki-nTree configuration files seem to be missing
```

If any additional information are needed I will send all necessary information. Hope that information make your app better.

pawcio0928, 2021-01-26:
Hello I resolve my issue. This is simple solution. On MacOS just browse to Applications/Python 3.6 and double-click Install Certificates.command. Should you add that information to the readme for other people?

eeintech, 2021-01-27:
@pawcio0928 Glad you solved your issue, is that required for all Python installations on MacOS? I will check this myself.

With the executable version, where you able to solve the `Warning: Failed to load` messages?

pawcio0928, 2021-01-27:
@eeintech I've already installed all dependency from InvenTree so I just run your installation script and thats it.

I have no idea how to solve problem with executable version.

eeintech, 2021-01-27:
@pawcio0928 For the executable version, it is most likely missing the configuration folders and files, they should be on the same root as the executable, do you see any? Else it means the `invoke build` command is not copying those and I would need to fix this.

pawcio0928, 2021-01-29:
@eeintech Please find below list of the folders and files in dist directory. It is looking ok.

```
kintree.tgz     kintree.zip     kintree_gui

config:
categories.yaml         digikey_categories.yaml inventree_dev.yaml      inventree_test.yaml     kicad_map.yaml          parameters_filters.yaml
digikey_api.yaml        digikey_parameters.yaml inventree_prod.yaml     kicad.yaml              parameters.yaml         version.yaml

kicad:
capacitor-polarized.dcm crystal-2p.lib          diode-standard.dcm      fuse.lib                oscillator-4p.dcm       resistor.lib            transistor-pnp.dcm
capacitor-polarized.lib default.dcm             diode-standard.lib      inductor.dcm            oscillator-4p.lib       transistor-nfet.dcm     transistor-pnp.lib
capacitor.dcm           default.lib             diode-zener.dcm         inductor.lib            protection-unidir.dcm   transistor-nfet.lib
capacitor.lib           diode-led.dcm           diode-zener.lib         integrated-circuit.dcm  protection-unidir.lib   transistor-npn.dcm
connector.dcm           diode-led.lib           ferrite-bead.dcm        integrated-circuit.lib  resistor-sm.dcm         transistor-npn.lib
connector.lib           diode-schottky.dcm      ferrite-bead.lib        library_template.dcm    resistor-sm.lib         transistor-pfet.dcm
crystal-2p.dcm          diode-schottky.lib      fuse.dcm                library_template.lib    resistor.dcm            transistor-pfet.lib
```

eeintech, 2021-01-29:
@pawcio0928 Yes indeed! I must check why then... time to give a macbook a shot...

eeintech, 2021-02-08:
Hello @pawcio0928 

I just gave Ki-nTree a shot on MacOS Mojave, and I'm at lost regarding all the issues I encountered...
First the GUI is missing the menu bar:

[image]

Without it, users can't configure the Digi-Key and InvenTree API endpoints... Same issue with setting KiCad's paths.
Does it happen to you too?

Looks like it could be related to PySimpleGUI not fully compatible with MacOS: https://github.com/PySimpleGUI/PySimpleGUI/issues/2624

Second thing I tried was to compile and run the executable version and am getting this error:
```
$ dist/kintree_gui 
Python path configuration:
  PYTHONHOME = '/var/folders/kh/ng9ls7td2034_r2t1c1cffv80000gn/T/_MEIIkDz9y'
  PYTHONPATH = (not set)
  program name = '/Users/francois/Desktop/scripts/Ki-nTree/Ki-nTree/dist/kintree_gui'
  isolated = 0
  environment = 0
  user site = 0
  import site = 0
  sys._base_executable = '/Users/francois/Desktop/scripts/Ki-nTree/Ki-nTree/dist/kintree_gui'
  sys.base_prefix = ''
  sys.base_exec_prefix = ''
  sys.platlibdir = 'lib'
  sys.executable = '/Users/francois/Desktop/scripts/Ki-nTree/Ki-nTree/dist/kintree_gui'
  sys.prefix = ''
  sys.exec_prefix = ''
  sys.path = [
    '/var/folders/kh/ng9ls7td2034_r2t1c1cffv80000gn/T/_MEIIkDz9y/base_library.zip',
    '/var/folders/kh/ng9ls7td2034_r2t1c1cffv80000gn/T/_MEIIkDz9y',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
LookupError: unknown encoding: utf-8

Current thread 0x00000001048935c0 (most recent call first):
<no Python frame>
```

I have very little knowledge of MacOS and unfortunately currently don't have more time to investigate this error and provide support for Mac users. For the time being I will say that Ki-nTree is solely compatible on Linux and Windows :)
Sorry for the trouble...

pawcio0928, 2021-02-09:
@eeintech Menu bar is displaying in my case. Please keep in mind that in Mac OS menu bar is on the top of the screen. 
[image]
 
I don't test Kicad functionality yet but Inventree works great. 

I don't manage to fix the executable version.

eeintech, 2021-02-09:
@pawcio0928 Oh you're right! I'm so not used to of MacOS way to do things 😬 
And glad to hear you can connect to InvenTree!

I'll try to figure out the executable situation in the future, could be a `pyinstaller` compatibility issue.

eeintech, 2021-02-14:
@pawcio0928 I was able to reproduce the same error when running the executable version:

``` bash
$ dist/kintree_gui 
[INFO]	Warning: Failed to load Digi-Key configuration
[INFO]	Warning: Failed to load InvenTree configuration
[INFO]	Warning: Failed to load KiCad configuration
[ERROR] File /Users/francois/Desktop/scripts/Ki-nTree/Ki-nTree/dist/config/user_files/kicad.yaml does not exists!
[INFO]	Warning: Failed to load Digi-Key configuration
[INFO]	Warning: Failed to load InvenTree configuration
[INFO]	Warning: Failed to load KiCad configuration

[ERROR]	Some Ki-nTree configuration files seem to be missing
```

Which is strange as all configuration files are present.

To fix the pyinstaller error:
``` bash
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
LookupError: unknown encoding: utf-8
```

I had to (pip) uninstall the `typing` package, references:
https://stackoverflow.com/questions/55833509/attributeerror-type-object-callable-has-no-attribute-abc-registry
https://github.com/pyinstaller/pyinstaller/issues/5438

eeintech, 2021-03-10:
I don't think I will have time to investigate more as I have very scarce access to MacOS, at least the main running option (eg. from command line) is working which is sufficient for the time being. Closing :smiley:

closed 2021-03-10 by eeintech