sparkmicro/Ki-nTree #16: invoke build on Windows

klemmma, 2021-02-13:
Hi,
I build Ki-nTree under Linux and Windows. However, on Windows I was struggeling:

`$ invoke build`

1st) xargs not found --> I had to add C:\Program Files\Git\usr\bin to PATH

2nd) wrong syntax
```
 ←[95m[MAIN]     Cleaning project directory←[0m
FIND: Parameter format not correct
rm: missing operand
```
by commenting a line in tasks.py, I got a quick workaround for me (see git diff at the bottom).

3rd) python error:
`AttributeError: type object 'Callable' has no attribute '_abc_registry'`
uninstalling typing did the job.
`pip uninstall typing`

With those fixes, it did work for me.
(Sorry, if I am inexact/on the wrong place here; it's my first open source bug report).

Hope that information may help you or someone.
Thanks for sharing the Ki-nTree!
Markus

```
diff --git a/tasks.py b/tasks.py
index 926ac2b..7af5e94 100644
--- a/tasks.py
+++ b/tasks.py
@@ -30,7 +30,7 @@ def update(c):
 @task
 def clean(c):
        cprint('[MAIN]\tCleaning project directory')
-       c.run('find . -name __pycache__ | xargs rm -r')
+       #c.run('find . -name __pycache__ | xargs rm -r')
        try:
                c.run('rm .coverage', hide='err')
        except UnexpectedExit:

```

eeintech, 2021-02-14:
@klemmma Thanks for the bug report! I swear I did test the build on Windows and had not found any problem... I will check again 😃

eeintech, 2021-02-14:
Oh I think I know why. You are not using Git Bash, are you?

I have tested only with Git Bash (Git for Windows) on Windows 10 and Python 3.8, inside a virtual environment.

Could you please describe your setup and method here so I can try to reproduce?

Thanks!

klemmma, 2021-02-15:
@eeintech I am not sure, what the rrot cause ist (in terms of configuration).
I use on Windows a basic Win10 with installed git (just reinstalled git for reproduction of the error):
1st) xargs
I have to add 'C:\Program Files\Git\usr\bin' to path. otherwise neither the git bash nor the windows cmd find xargs
2nd) find: I do get the same error no matter of I run it in git bash or in windows cmd: ```
```

C:\Users\test\kicad\Ki-nTree>find . -name __pycache__ | xargs rm -r
FIND: Parameter format not correct
rm: missing operand
Try 'rm --help' for more information.

C:\Users\test\kicad\Ki-nTree>bash

test@Win-PC  ~/kicad/Ki-nTree (master)
$ find . -name __pycache__ | xargs rm -r
FIND: Parameter format not correct
rm: missing operand
Try 'rm --help' for more information.
```

When I check for find in bash, it seems to fallback into the native Windows find for which the syntax of the command is not right:

```
test@Win-PC  ~/kicad/Ki-nTree (master)
$ find /?
Searches for a text string in a file or files.

FIND [/V] [/C] [/N] [/I] [/OFF[LINE]] "string" [[drive:][path]filename[ ...]]

  /V         Displays all lines NOT containing the specified string.
  /C         Displays only the count of lines containing the string.
  /N         Displays line numbers with the displayed lines.
  /I         Ignores the case of characters when searching for the string.
  /OFF[LINE] Do not skip files with offline attribute set.
  "string"   Specifies the text string to find.
  [drive:][path]filename
             Specifies a file or files to search.

If a path is not specified, FIND searches the text typed at the prompt
or piped from another command.

test@Win-PC  ~/kicad/Ki-nTree (master)
```
https://docs.microsoft.com/en-us/windows-server/administration/windows-commands/find

eeintech, 2021-02-15:
@klemmma Yes the `find` command is very different in Windows shell vs Linux shell.

What I recommend for you is to [download and install Git Bash](https://gitforwindows.org/), and this is what you would get for the `find` command:

[image]

Therefore all the `invoke` command would be supported 😃

klemmma, 2021-02-15:
Ah, I see the difference (I think):
You are using MinGW, which is a difference to pure git bash...
in git bash, I can use the windows-commands as well, as the paths are still existing:
```
C:\>where bash
C:\Program Files\Git\usr\bin\bash.exe
C:\>bash

test@Win-PC  ~
$ where find
C:\Windows\System32\find.exe
C:\Program Files\Git\usr\bin\find.exe

test@Win-PC  ~
$ find
FIND: Parameter format not correct
```
the find command in git bash still points to the windows find, as it is the first in my path list.
in git-bash:
```
test@Win-PC  ~/kicad/Ki-nTree (master)
$ "C:/Program Files/Git/usr/bin/find.exe" . -name __pycache__ | xargs rm -r
```
does the job (for me).

eeintech, 2021-02-15:
Sweet! Yes that's probably it, I don't really use Windows much so I trust your feedback :smile: 

I think you should be able to run the `invoke exec` without issue, because it does call `PyInstaller` only.
Does the `invoke copy-configuration` command works?

eeintech, 2021-03-04:
@klemmma Any more work on this or can I close it?

klemmma, 2021-03-05:
@eeintech sorry, did not follow this any more, as it is working for me already.

closed 2021-03-05 by klemmma

eeintech, 2021-03-06:
Sweet thanks!