sparkmicro/Ki-nTree #19: "Proceed to localhost (unsafe)" not available in Google Chrome

cdwilson, 2021-03-16:
I was following the directions at https://github.com/sparkmicro/Ki-nTree/#get-digi-key-api-token but Chrome did not give me the option to "Proceed to localhost (unsafe)".

I was able to finally get this working by setting this flag in Chrome `chrome://flags/#allow-insecure-localhost`:

[image]

Also, when registering an app on the Digi-Key website, the OAuth Callback needs to be set to `https://localhost:8139/digikey_callback`

eeintech, 2021-03-16:
> I was able to finally get this working by setting this flag in Chrome `chrome://flags/#allow-insecure-localhost`:

This must be new to Chrome... Added to #21, thanks for the feedback! 

> Also, when registering an app on the Digi-Key website, the OAuth Callback needs to be set to `https://localhost:8139/digikey_callback`

Does it really? I have it set to `https://localhost` and never had any issue...

cdwilson, 2021-03-16:
> Does it really? I have it set to https://localhost and never had any issue...

You're right, `https://localhost` might be OK.  When I was trying to fix the OAuth issues, I was just following the directions from https://github.com/peeter123/digikey-api/blob/master/README.md#register which say

> When registering an app the OAuth Callback needs to be set to https://localhost:8139/digikey_callback.

closed 2021-03-16 by cdwilson