sparkmicro/Ki-nTree #20: Error: Symbol libraries were not found

cdwilson, 2021-03-16:
I'm trying to get Ki-nTree working for the first time, but I'm getting the following error when I click "Submit":

[image]

[image]

Here is my `user_files/kicad.yaml`:
```
KICAD_ENABLE: true
KICAD_FOOTPRINTS_PATH: /Users/chris/KiCad/Ki-nTree/footprints/
KICAD_SYMBOLS_PATH: /Users/chris/KiCad/Ki-nTree/symbols/
KICAD_TEMPLATES_PATH: /Users/chris/KiCad/Ki-nTree/templates/
```

These folders exist on my system:
```
$ ls /Users/chris/KiCad/Ki-nTree/
footprints symbols    templates
```

Any suggestions for how to figure out what's going wrong?

cdwilson, 2021-03-16:
I didn't realize that the library files already need to exist in these directories (they were empty).  I added empty KiCad library files into these folders and I fixed the `KICAD_TEMPLATES_PATH` to point to the Ki-nTree template folder.

[image]

```
$ ls ~/KiCad/Ki-nTree/{footprints,symbols}
/Users/chris/KiCad/Ki-nTree/footprints:
kintree.pretty

/Users/chris/KiCad/Ki-nTree/symbols:
kintree.dcm kintree.lib
```

When I clicked "Submit" button, I get the following error:

```
$ python kintree_gui.py 
/Users/chris/.anyenv/envs/pyenv/versions/inventree/lib/python3.9/site-packages/PySimpleGUI/PySimpleGUI.py:19431: UserWarning: You are running a VERY old version of tkinter 8.5.9
  warnings.warn('You are running a VERY old version of tkinter {}'.format(tclversion_detailed), UserWarning)
DEPRECATION WARNING: The system version of Tk is deprecated and may be removed in a future release. Please don't rely on it. Set TK_SILENCE_DEPRECATION=1 to suppress this warning.

[MAIN]  Connecting to Inventree server
[TREE]  Successfully connected to InvenTree server (ENV=TESTING)

--------------------

[MAIN]  Using Digi-Key cached data for 311-0.0LRCT-ND

[MAIN]  Creating part in Inventree
[INFO]  Category: "Resistors"
[INFO]  Subcategory: "Surface Mount"
[INFO]  User Category: "Resistors"
[INFO]  User Subcategory: "Surface Mount"
{
    "Resistors": {
        "kintree": "/Users/chris/KiCad/Ki-nTree/symbols/kintree.lib"
    }
}
Traceback (most recent call last):
  File "/Users/chris/Projects/Ki-nTree/kintree_gui.py", line 1128, in <module>
    main()
  File "/Users/chris/Projects/Ki-nTree/kintree_gui.py", line 959, in main
    symbol, template, footprint = user_defined_symbol_template_footprint(categories, part_info['manufacturer_part_number'])
  File "/Users/chris/Projects/Ki-nTree/kintree_gui.py", line 613, in user_defined_symbol_template_footprint
    footprint_library = config_interface.load_footprint_paths(user_config_path=settings.CONFIG_KICAD_CATEGORY_MAP,
  File "/Users/chris/Projects/Ki-nTree/config/config_interface.py", line 245, in load_footprint_paths
    footprint_libraries_paths['uncategorized'])
KeyError: 'uncategorized'
```

In addition, now when I try to start Ki-nTree, I get the same error when trying to start the app:

```
$ python kintree_gui.py 
Traceback (most recent call last):
  File "/Users/chris/Projects/Ki-nTree/kintree_gui.py", line 6, in <module>
    import config.settings as settings
  File "/Users/chris/Projects/Ki-nTree/config/settings.py", line 158, in <module>
    footprint_libraries_paths = config_interface.load_footprint_paths(
  File "/Users/chris/Projects/Ki-nTree/config/config_interface.py", line 245, in load_footprint_paths
    footprint_libraries_paths['uncategorized'])
KeyError: 'uncategorized'
```

eeintech, 2021-03-16:
@cdwilson Good catch, it doesn't like the fact that your `/Users/chris/KiCad/Ki-nTree/footprints/kintree.pretty/` folder is empty.

Change `config/user_files/kicad.yaml` from:
``` yaml
KICAD_FOOTPRINTS_PATH: /Users/chris/KiCad/Ki-nTree/footprints/kintree.pretty/
```

to:
``` yaml
KICAD_FOOTPRINTS_PATH: /Users/chris/KiCad/Ki-nTree/footprints/
```

And that should fix it. The footprint directory should be the base of all KiCad footprint libraries, which are folders.

I have fixed this crash/bug in #21 so it does not happen to others :wink: 

Btw, the template folder path can be relative (eg. `kicad/templates/`) which I believe is the default value.

eeintech, 2021-03-23:
I think we can close this one, should be fixed in `0.3.5` :smiley:

closed 2021-03-23 by eeintech