sparkmicro/Ki-nTree #23: Crash when clicking "Check SnapEDA" button

cdwilson, 2021-03-16:
[image]

```
Traceback (most recent call last):
  File "/Users/chris/Projects/Ki-nTree/kintree_gui.py", line 1128, in <module>
    main()
  File "/Users/chris/Projects/Ki-nTree/kintree_gui.py", line 959, in main
    symbol, template, footprint = user_defined_symbol_template_footprint(categories, part_info['manufacturer_part_number'])
  File "/Users/chris/Projects/Ki-nTree/kintree_gui.py", line 704, in user_defined_symbol_template_footprint
    snapeda_window(part_number)
  File "/Users/chris/Projects/Ki-nTree/kintree_gui.py", line 263, in snapeda_window
    snapeda_event, snapeda_values = snapeda_window.read()
  File "/Users/chris/.anyenv/envs/pyenv/versions/inventree/lib/python3.9/site-packages/PySimpleGUI/PySimpleGUI.py", line 8200, in read
    results = self._read(timeout=timeout, timeout_key=timeout_key)
  File "/Users/chris/.anyenv/envs/pyenv/versions/inventree/lib/python3.9/site-packages/PySimpleGUI/PySimpleGUI.py", line 8266, in _read
    self._Show()
  File "/Users/chris/.anyenv/envs/pyenv/versions/inventree/lib/python3.9/site-packages/PySimpleGUI/PySimpleGUI.py", line 8027, in _Show
    StartupTK(self)
  File "/Users/chris/.anyenv/envs/pyenv/versions/inventree/lib/python3.9/site-packages/PySimpleGUI/PySimpleGUI.py", line 13704, in StartupTK
    _convert_window_to_tk(window)
  File "/Users/chris/.anyenv/envs/pyenv/versions/inventree/lib/python3.9/site-packages/PySimpleGUI/PySimpleGUI.py", line 13576, in _convert_window_to_tk
    PackFormIntoFrame(window, master, window)
  File "/Users/chris/.anyenv/envs/pyenv/versions/inventree/lib/python3.9/site-packages/PySimpleGUI/PySimpleGUI.py", line 12917, in PackFormIntoFrame
    photo = tk.PhotoImage(file=element.Filename)
  File "/Users/chris/.anyenv/envs/pyenv/versions/3.9.1/lib/python3.9/tkinter/__init__.py", line 4062, in __init__
    Image.__init__(self, 'photo', name, cnf, master, **kw)
  File "/Users/chris/.anyenv/envs/pyenv/versions/3.9.1/lib/python3.9/tkinter/__init__.py", line 4007, in __init__
    self.tk.call(('image', 'create', imgtype, name,) + options)
_tkinter.TclError: couldn't recognize data in image file "/Users/chris/Projects/Ki-nTree/search/images/rc0402fr-07330rl_symbol.png"
```

eeintech, 2021-03-16:
@cdwilson Can you let me know the part number with which you encountered this crash?

cdwilson, 2021-03-16:
311-330LRCT-ND (Yageo RC0402FR-07330RL)

(if this happens in the future, sometimes you may be able to get a hint from the name of the file in the trace `rc0402fr-07330rl_symbol.png`)

eeintech, 2021-03-17:
@cdwilson Unfortunately I'm not able to reproduce this issue... 

Can you do me a favor and copy-paste this code in a new file:
``` python
import PySimpleGUI as sg

layout = [[sg.Image(r'/Users/chris/Projects/Ki-nTree/search/images/rc0402fr-07330rl_symbol.png')]]

mainWin = sg.Window('Main Window',resizable=True).Layout(layout)

while True:
  event, values = mainWin.Read(timeout=0)
```

then run it. Does it crash the same way?

cdwilson, 2021-03-17:
Yes, it crashes the same way.  I wonder if it's a tkinter issue (I get a tkinter warning whenever I run `kintree_gui.py`).  Let me see if there is a way to upgrade tkinter to a newer version and see if that fixes it.  What tkinter version are you using?

```
$ python test_23.py 
/Users/chris/.anyenv/envs/pyenv/versions/inventree/lib/python3.9/site-packages/PySimpleGUI/PySimpleGUI.py:19431: UserWarning: You are running a VERY old version of tkinter 8.5.9
  warnings.warn('You are running a VERY old version of tkinter {}'.format(tclversion_detailed), UserWarning)
DEPRECATION WARNING: The system version of Tk is deprecated and may be removed in a future release. Please don't rely on it. Set TK_SILENCE_DEPRECATION=1 to suppress this warning.
Traceback (most recent call last):
  File "/Users/chris/Projects/Ki-nTree/test_23.py", line 8, in <module>
    event, values = mainWin.Read(timeout=0)
  File "/Users/chris/.anyenv/envs/pyenv/versions/inventree/lib/python3.9/site-packages/PySimpleGUI/PySimpleGUI.py", line 8200, in read
    results = self._read(timeout=timeout, timeout_key=timeout_key)
  File "/Users/chris/.anyenv/envs/pyenv/versions/inventree/lib/python3.9/site-packages/PySimpleGUI/PySimpleGUI.py", line 8250, in _read
    event, values = self._ReadNonBlocking()
  File "/Users/chris/.anyenv/envs/pyenv/versions/inventree/lib/python3.9/site-packages/PySimpleGUI/PySimpleGUI.py", line 8377, in _ReadNonBlocking
    self._Show(non_blocking=True)
  File "/Users/chris/.anyenv/envs/pyenv/versions/inventree/lib/python3.9/site-packages/PySimpleGUI/PySimpleGUI.py", line 8027, in _Show
    StartupTK(self)
  File "/Users/chris/.anyenv/envs/pyenv/versions/inventree/lib/python3.9/site-packages/PySimpleGUI/PySimpleGUI.py", line 13704, in StartupTK
    _convert_window_to_tk(window)
  File "/Users/chris/.anyenv/envs/pyenv/versions/inventree/lib/python3.9/site-packages/PySimpleGUI/PySimpleGUI.py", line 13576, in _convert_window_to_tk
    PackFormIntoFrame(window, master, window)
  File "/Users/chris/.anyenv/envs/pyenv/versions/inventree/lib/python3.9/site-packages/PySimpleGUI/PySimpleGUI.py", line 12917, in PackFormIntoFrame
    photo = tk.PhotoImage(file=element.Filename)
  File "/Users/chris/.anyenv/envs/pyenv/versions/3.9.1/lib/python3.9/tkinter/__init__.py", line 4062, in __init__
    Image.__init__(self, 'photo', name, cnf, master, **kw)
  File "/Users/chris/.anyenv/envs/pyenv/versions/3.9.1/lib/python3.9/tkinter/__init__.py", line 4007, in __init__
    self.tk.call(('image', 'create', imgtype, name,) + options)
_tkinter.TclError: couldn't recognize data in image file "/Users/chris/Projects/Ki-nTree/search/images/rc0402fr-07330rl_symbol.png"
```

eeintech, 2021-03-17:
@cdwilson This is what I have, from inside my virtual env:
``` bash
$ python -c "import tkinter; print(tkinter.TkVersion)"
8.6
```

Update Tkinter then re-try?

cdwilson, 2021-03-17:
Yeah, the crash was due to an old version of tkinter (8.5.9) shipped with Mac...

I use [pyenv](https://github.com/pyenv/pyenv) to manage python version installations (and I use [anyenv](https://github.com/anyenv/anyenv) to manage installation of pyenv, rbenv, etc.).  I think pyenv just grabs the system installed tcl/tk framework which is pretty old.

After rebuilding python against newer tcl-tk installed via homebrew, everything works as expected, and I see the SnapEDA image:

[image]

Here are the steps I used to rebuild python:

```
$ brew install tcl-tk
$ env \
  PATH="$(brew --prefix tcl-tk)/bin:$PATH" \
  LDFLAGS="-L$(brew --prefix tcl-tk)/lib" \
  CPPFLAGS="-I$(brew --prefix tcl-tk)/include" \
  PKG_CONFIG_PATH="$(brew --prefix tcl-tk)/lib/pkgconfig" \
  CFLAGS="-I$(brew --prefix tcl-tk)/include" \
  PYTHON_CONFIGURE_OPTS="--with-tcltk-includes='-I$(brew --prefix tcl-tk)/include' --with-tcltk-libs='-L$(brew --prefix tcl-tk)/lib -ltcl8.6 -ltk8.6'" \
  pyenv install 3.9.1
$ python
Python 3.9.1 (default, Mar 17 2021, 07:34:15) 
[Clang 12.0.0 (clang-1200.0.32.29)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import tkinter
>>> tkinter.TclVersion, tkinter.TkVersion
(8.6, 8.6)
>>> tkinter._test()
```

eeintech, 2021-03-17:
Sweet, should work now :smiley:

closed 2021-03-17 by cdwilson