sparkmicro/Ki-nTree #25: ConnectionResetError: [Errno 104] Connection reset by peer

zinka, 2021-03-20:
Hi,

I have been able to setup an InvenTree server on AWS and able to connect to it through Ki-ntree (V.0.3.2). I am also able to use Digikey's API ID. InvenTree details are given below:

[image]

The problem is ... once in a while I am able to create parts using Ki-nTree ... but "most of the times" when I try to create a part through Ki-nTree, I am getting the below error (sorry for the lengthy error message):

`ConnectionResetError: [Errno 104] Connection reset by peer

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/zinka/Dropbox/HCS/Enginering/Ki-nTree/env-kintree/lib/python3.8/site-packages/requests/models.py", line 753, in generate
    for chunk in self.raw.stream(chunk_size, decode_content=True):
  File "/home/zinka/Dropbox/HCS/Enginering/Ki-nTree/env-kintree/lib/python3.8/site-packages/urllib3/response.py", line 576, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/home/zinka/Dropbox/HCS/Enginering/Ki-nTree/env-kintree/lib/python3.8/site-packages/urllib3/response.py", line 541, in read
    raise IncompleteRead(self._fp_bytes_read, self.length_remaining)
  File "/usr/lib/python3.8/contextlib.py", line 131, in __exit__
    self.gen.throw(type, value, traceback)
  File "/home/zinka/Dropbox/HCS/Enginering/Ki-nTree/env-kintree/lib/python3.8/site-packages/urllib3/response.py", line 455, in _error_catcher
    raise ProtocolError("Connection broken: %r" % e, e)
urllib3.exceptions.ProtocolError: ("Connection broken: ConnectionResetError(104, 'Connection reset by peer')", ConnectionResetError(104, 'Connection reset by peer'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/zinka/Dropbox/HCS/Enginering/Ki-nTree/kintree_gui.py", line 1106, in <module>
    main()
  File "/home/zinka/Dropbox/HCS/Enginering/Ki-nTree/kintree_gui.py", line 951, in main
    new_part, part_pk, part_data = inventree_interface.inventree_create(part_info=part_info,
  File "/home/zinka/Dropbox/HCS/Enginering/Ki-nTree/database/inventree_interface.py", line 306, in inventree_create
    subcategory_pk = inventree_api.get_inventree_category_id(category_name=subcategory_name,
  File "/home/zinka/Dropbox/HCS/Enginering/Ki-nTree/database/inventree_api.py", line 36, in get_inventree_category_id
    part_categories = PartCategory.list(inventree_api)
  File "/home/zinka/Dropbox/HCS/Enginering/Ki-nTree/database/inventree-python/inventree/base.py", line 69, in list
    response = api.get(cls.URL, params=params, **kwargs)
  File "/home/zinka/Dropbox/HCS/Enginering/Ki-nTree/database/inventree-python/inventree/api.py", line 348, in get
    response = self.request(url, method='get', **kwargs)
  File "/home/zinka/Dropbox/HCS/Enginering/Ki-nTree/database/inventree-python/inventree/api.py", line 202, in request
    response = methods[method](
  File "/home/zinka/Dropbox/HCS/Enginering/Ki-nTree/env-kintree/lib/python3.8/site-packages/requests/api.py", line 76, in get
    return request('get', url, params=params, **kwargs)
  File "/home/zinka/Dropbox/HCS/Enginering/Ki-nTree/env-kintree/lib/python3.8/site-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/zinka/Dropbox/HCS/Enginering/Ki-nTree/env-kintree/lib/python3.8/site-packages/requests/sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/zinka/Dropbox/HCS/Enginering/Ki-nTree/env-kintree/lib/python3.8/site-packages/requests/sessions.py", line 697, in send
    r.content
  File "/home/zinka/Dropbox/HCS/Enginering/Ki-nTree/env-kintree/lib/python3.8/site-packages/requests/models.py", line 831, in content
    self._content = b''.join(self.iter_content(CONTENT_CHUNK_SIZE)) or b''
  File "/home/zinka/Dropbox/HCS/Enginering/Ki-nTree/env-kintree/lib/python3.8/site-packages/requests/models.py", line 756, in generate
    raise ChunkedEncodingError(e)
requests.exceptions.ChunkedEncodingError: ("Connection broken: ConnectionResetError(104, 'Connection reset by peer')", ConnectionResetError(104, 'Connection reset by peer'))`

Could you please let me know if it possible to rectify this issue? Apart from Ki-nTree,  don't have any issues in using InvenTree itself through a browser.

eeintech, 2021-03-22:
Hello @zinka 

It looks like as if the InvenTree API connection drops between you hitting "CREATE" button and the start of part creation...
I'm not quite sure what's happening.

Do you mind updating to `0.3.4` and re-try to see if this issue is still hapenning?
I have made few updates to the part creation flow in `0.3.3` release.

Thanks!

zinka, 2021-04-02:
Hi @eeintech

Sorry for the late reply. I have tried the latest version of Ki-nTree (0.3.5). Still facing the same issue. This doesn't happen when I try to add a part to a local(host) InvenTree database. So, I will add the part to local database first ... and then will synchronise the remote data using backup&restore features of InvenTree.

Thank you.

eeintech, 2021-04-02:
This looks like a problem with the connection to your external InvenTree instance. Are you running the latest version of InvenTree?

eeintech, 2021-04-07:
@zinka Still having this problem updating your remote instance of InvenTree?

eeintech, 2021-04-13:
@zinka I closed this issue because I don't think it is related to Ki-nTree but that the InvenTree remote instance was quite old. If you have any other issue after updating InvenTree let me know :smiley:

closed 2021-04-13 by eeintech