sparkmicro/Ki-nTree #27: Automatically creating categories and parameters in InvenTree

cdwilson, 2021-03-22:
When I add a new new part type in Ki-nTree, I get the following errors if the category and parameters don't already exist in InvenTree:

```
[MAIN]  Creating part in Inventree
[INFO]  Category: "Mechanicals"
[INFO]  Subcategory: "Switch"
[INFO]  User Category: "Mechanicals"
[INFO]  User Subcategory: "Switch"
[INFO]  Warning: Parameter "Between Board Height" not found in supplier data
[INFO]  Warning: Parameter "Screw, Thread Size" not found in supplier data

[TREE]  Warning: Subcategory "Switch" does not exist
[ERROR] Categories (Mechanicals - Switch) does not exist in InvenTree
```

```
[MAIN]  Creating parameters
[TREE]  Error: Parameter template "Interface Type" does not exist
[TREE]  Error: Parameter template "Locking" does not exist
[TREE]  Error: Parameter template "Mating Height" does not exist
[TREE]  Error: Parameter template "Mounting Type" does not exist
[TREE]  Error: Parameter template "Number of Contacts" does not exist
[TREE]  Error: Parameter template "Number of Rows" does not exist
[TREE]  Error: Parameter template "Package Height" does not exist
[TREE]  Error: Parameter template "Pitch" does not exist
[TREE]  Error: Parameter template "Temperature Range" does not exist
[TREE]  Error: Parameter template "Value" does not exist
[TREE]  Error: Parameter template "Frequency" does not exist
[TREE]  Error: Parameter template "Orientation" does not exist
[TREE]  Error: Parameter template "Polarity" does not exist
[TREE]  Error: Parameter template "Shielding" does not exist
[TREE]  Error: Parameter template "Symbol" does not exist
[TREE]  Error: Parameter template "Footprint" does not exist
[TREE]  Warning: The following parameters were skipped:
--->    Interface Type
--->    Locking
--->    Mating Height
--->    Mounting Type
--->    Number of Contacts
--->    Number of Rows
--->    Package Height
--->    Pitch
--->    Temperature Range
--->    Value
--->    Frequency
--->    Orientation
--->    Polarity
--->    Shielding
--->    Symbol
--->    Footprint
```

I can create these manually in InvenTree (which is what I've been doing for Category), but I'm wondering if there is a more "automatic" way to populate these parameters into InvenTree so that I don't have to add them manually each time I add a new part type.

cdwilson, 2021-03-22:
Whoops...nevermind!  I totally missed the `invoke setup-inventree` task... :man_facepalming:

closed 2021-03-22 by cdwilson

eeintech, 2021-03-22:
@cdwilson My fault... documentation is lacking... Good to know you found it :wink: