sparkmicro/Ki-nTree #29: Configuration options for part data

cdwilson, 2021-04-02:
@eeintech just wanted to let you know about a couple minor changes I've hacked into my local copy of to Ki-nTree to tailor it for my own usage, and see if you'd be interested in a PR to make them into configuration options in the settings.

1. For part description, I use the `detailed_description` from Digi-Key instead of `product_description`
2. For keywords, I use the `product_description` instead of `inventree_part['description']`
3. For datasheet, I use the `product_url` since this ends up being the URL on the `SupplierPart` page in InvenTree.

[image]

Also, once inventree/InvenTree#1417 gets merged in, I'd like to add the ability for Ki-nTree to push both the `primary_datasheet` into the `ManufacturerPart` and the `product_url` into the `SupplierPart`.  

It would be ideal if the `ManufacturerPart` in InvenTree eventually supports separate URLs for the Manufacturer's product page vs. the datasheet so that we can have both links available on the MPN.  For example, if we look at [497-11767-ND](https://www.digikey.com/short/pjmp0j49):

`ManufacturerPart`
- Manufacturer's product URL: https://www.st.com/en/microcontrollers-microprocessors/stm32f405rg.html
- Manufacturer's datasheet URL (`primary_datasheet`): https://www.st.com/content/ccc/resource/technical/document/datasheet/ef/92/76/6d/bb/c2/4f/f7/DM00037051.pdf/files/DM00037051.pdf/jcr:content/translations/en.DM00037051.pdf 

`SupplierPart`
- Supplier's product URL (`product_url`): https://www.digikey.com/product-detail/en/stmicroelectronics/STM32F405RGT6/497-11767-ND/2754208

eeintech, 2021-04-02:
@cdwilson Looks like a good idea!

If you do send a PR for flexible Digi-Key configuration, I would put it under a new file in "config > digikey > digikey_config.yaml".
And make sure the default is what's in right now :smiley: 

Once we complete the implementation of ManufacturerPart, I'll update the inventree-python API and Ki-nTree :+1:

cdwilson, 2021-04-02:
> If you do send a PR for flexible Digi-Key configuration, I would put it under a new file in "config > digikey > digikey_config.yaml".
> And make sure the default is what's in right now 😃

Yeah, I'll make sure to do that.  I also realized that we probably want a similar yaml configuration file for some of the other user-configurable settings that exist in the `settings.py` file today (like the IPN changes I made in #26).

> Once we complete the implementation of ManufacturerPart, I'll update the inventree-python API and Ki-nTree 👍

Great!  I may wait on this PR until after you make these changes so I can add an option to push both the `primary_datasheet` and the `product_url` into the respective `ManufacturerPart` / `SupplierPart`.

eeintech, 2021-04-07:
> Yeah, I'll make sure to do that. I also realized that we probably want a similar yaml configuration file for some of the other user-configurable settings that exist in the `settings.py` file today (like the IPN changes I made in #26).

For that I started PR #31. I will also add a bunch of general settings for controlling the software behavior.

eeintech, 2021-04-07:
@cdwilson I updated the PR to handle the three Digi-Key settings from your original request, try it out! :smiley:

PR #31 (2021-04-13): 0.3.6 - Improved Settings
- General settings are now stored in `config/settings/general.yaml`
  - InvenTree environment setting can now be loaded from this file and overwritten with environmental variable (if necessary)
- Internal Part Number settings from #26 are now stored in `config/settings/internal_part_number.yaml`
- Digi-Key search settings are now stored in `config/digikey/digikey_config.yaml`
  - Added integration for settings requested by @cdwilson, fixes #29 
- Updated settings to load from files instead of hard-coded
closed 2021-04-13 by eeintech