sparkmicro/Ki-nTree #32: Turning this into a Python package?

randrej, 2021-04-11:
If I were to try turning this into a Python package that can be pushed to PyPI or at least installed via pip, would this be a welcome changeset? I'd make a pyproject.toml and setup.py.

How would we benefit from this? I could then make a package in Arch Linux's AUR, other people could package it for other distros, a large number of people could even just do a pip install and the package would run.

eeintech, 2021-04-12:
@randrej If you could do that, it would be amazing! I did not dig enough on that side when I looked, it would be awesome if Ki-nTree came in a more friendly packaging.

I was also interested into [BeeWare](https://beeware.org/) but never had the chance in trying it.

randrej, 2021-04-13:
Okay, I have managed to do a large part of what's necessary.

Config folder path has to be changed. If you install the package from pip or your Linux package manager, the configuration files can't be located relative to sources like they are currently, they have to be stored in user-specific places on any OS. This is surprisingly complex, but we can offload config folder location duties to [appdirs](https://pypi.org/project/appdirs/). We keep the same config structure, we just obtain the destination dir from `appdirs`.

To be clear, the way we're storing configs is idiomatic for Linux, but not so for Windows. Most Windows applications store settings in the registry. We could overhaul the settings to respect these OS-dependent conventions by incorporating something like [profig](https://pypi.org/project/profig/) or QSettings in PySide / PyQt5 (though this would only make sense if the project used Qt), but this isn't an easy solution. First off, all of the settings would have to be editable in the application, which is inconvenient - it's easier to edit them in an editor. Second, most of the settings are list- or dict-based, and handling such structures in Windows Registry and .ini / .conf files isn't as convenient as editing a YAML file.

I suggest we just use `appdirs` to store the config files in appropriate locations, but keep yaml-based config files as the main way of configuring the application. This is the natural way of doing things in Linux, config files in ~/.config/kintree. Windows users would have to edit files at `'C:\Users\<username>\AppData\Local\Sparkmicro\KinTree'`

eeintech, 2021-04-13:
Wow thanks for digging up all this information, really appreciated :+1: 

> I suggest we just use `appdirs` to store the config files in appropriate locations, but keep yaml-based config files as the main way of configuring the application. This is the natural way of doing [...]

After looking up the `appdirs` package, I agree that this looks like a very adequate solution to storing configuration files. So go for it!

> After fixing this, I just have to add a bunch of tests to see if everything works, fix arising issues and that's it, you may post it to PyPI. I could then also add Github Actions for automatic pushing to PyPI, automatic linting, etc.

That's sounds **really** good, can't wait to see this new workflow! I was also thinking of moving the CI to Github Actions, I'll try to find some time to migrate soon.

eeintech, 2021-04-15:
@randrej Any chance I can try this out on my end?

randrej, 2021-04-15:
Yeah, I'll post a PR when I have some timem

randrej, 2021-04-17:
The PR is up. #35

randrej, 2021-04-18:
I'm looking at the tests, and I see that they're integration tests: they involve actually communicating with an Inventree instance and the Digikey API. While this is effective at testing whether the whole thing works, it's very cumbersome for testing specifics and the scope is huge.

I believe a better approach would be writing unit tests (with *pytest*) that only check the functionality of the Ki-nTree itself, with all external dependencies mocked.

Let me explain with an example:
We are testing the process of finding a part on Digikey, figuring out which category it should go into, and inserting it into InvenTree. We replace the actual Digikey API module with a mock one, which expects some specific calls to be made. We also have the mock module return some specific values. This deterministically simulates how this API behaves every time. Then, we do the same thing with the InvenTree API module. It expects some data, and returns some specific data. Now we can write the tests for the actual functionality of Ki-nTree itself - what happens when the part exists, what happens when it doesn't, etc.

In the end, it does make sense to retain the existing tests to test for API-side breakages by Digikey or Inventree, but I'd also clean these up and put them into PyTest. I propose I first port those integration tests you wrote into pytest, and we can go on from there.

A text on mocking: https://changhsinlee.com/pytest-mock/

eeintech, 2021-04-18:
@randrej Sorry I haven't had the chance to take a look at the PR yet, I will very soon! Looks very exciting!

I agree with you my test strategy was very rudimentary... Using `pytest` and unit tests instead would be great.
I read the article you sent (very clear) and with your example, I do agree that mocking does makes a lot of sense for testing Ki-nTree. I would love to see this in action 😃 

Thanks so much for contributing to improve Ki-nTree, it's a huge help and I'm learning quite a bit too!

closed 2021-07-15 by eeintech