sparkmicro/Ki-nTree #33: Code formatting, linting

randrej, 2021-04-13:
I propose adopting [`black`](https://github.com/psf/black) to format the source code in the project. This is a bit controversial, because the first commit after doing so would likely change every line in the project since the project uses tabs for indentation, contrary to pep8.

I also propose adding configs for mypy and flake8, and gradually adding types to the code. We could then use [`pre-commit`](https://pre-commit.com/) to run all these tools and make sure the code is formatted and linted before committing. Here's one of my packages as an example: [srtools](https://gitlab.com/andrejr/srtools).

eeintech, 2021-04-13:
@randrej I have tested `black` in the past and found out it is too opinionated, I felt like wasting my time fixing my code formatting than focusing on the code itself because of all those unexpected rules. Maybe I found it quite annoying because I don't have a background of software engineering.

However I do agree that some of the style should be checked (not the case now) and found that `flake8` was a good option while working on [InvenTree](https://github.com/inventree/inventree)'s codebase. I was going to give it a shot soon when I find some time, but if you do have some in your hands, don't hesitate!

randrej, 2021-04-13:
I like `black` not only because it's opinionated, but also because it's a formatter and not just a linter. I type in code without caring for proper formatting and then just trigger it in my editor. I do it in [n]vim, but every major editor has support for this, too. I also have `black` it among my pre-commit hooks, in case I forgot to run it on some file. In this way, it doesn't waste my time - it saves it.

eeintech, 2021-04-14:
@randrej Understood. However, I'd rather not use `black` for the moment, sorry. `flake8` should definitely run in the CI though.

cdwilson, 2021-04-15:
FWIW, I'd be in support of using `black` as a formatter together with `pre-commit`, but I totally understand this is one of those personal preference things.  @eeintech not intended in any way to try and persuade you differently (happy with your suggestion for `flake8`), just wanted to say that I'd be in favor of this proposal if you ever change your mind and want to go down this route.

eeintech, 2021-04-15:
@cdwilson Thanks for chiming in. I just remember only frustration using `black` because it was forcing me to format my code in one "narrow-minded" direction which I didn't like overall so I really don't want to start this again. Code should have consistent and nice formatting but I think of `black` as too many rules. PEP rules are enough in my opinion to keep a code base clean, as long as the developer are respectful to the original coding style. I believe that the code interpretation and reading ease depends on people, so one set of rules cannot rule them all.

eeintech, 2021-06-29:
I'm quite happy with PEP checks, the code base is now compliant :smiley:

closed 2021-06-29 by eeintech