sparkmicro/Ki-nTree #34: Add Supplier Link field

cdwilson, 2021-04-15:
Now that https://github.com/inventree/InvenTree/pull/1417 has been merged and InvenTree supports ManufacturerParts, I'm wondering if you would be open to adding a separate "Supplier Link" field which uses `product_url` as the default value:

[image]

This would map to the existing Supplier Link field in InvenTree:

[image]

The other requested change would be to change the existing Datasheet field to map to the new ManufacturerPart Link in InvenTree:

[image]

[image]

What do you think?

eeintech, 2021-04-15:
I think we should fully embrace the new ManufacturerPart model and add it to Ki-nTree :smile:

eeintech, 2021-04-30:
@cdwilson I added support for `ManufacturerPart` model, want to take a look? See #38

closed 2021-05-03 by eeintech

cdwilson, 2021-05-04:
@eeintech just tried out the new changes, looks great, thanks for adding this!

eeintech, 2021-05-04:
@cdwilson You're welcome! :+1:

PR #38 (2021-05-03): 0.3.7
- Added support for InvenTree `ManufacturerPart` model introduced in https://github.com/inventree/InvenTree/pull/1417
  - Decoupled manufacturer and supplier parts creation
  - Updated InvenTree API package
  - Added supplier's website link field in part form (fixes #34)
  - Re-ordered manufacturer and supplier fields in part form

[image]

- Fixed Ferrite Bead template reference designator letter (`FB` instead of `L`)