sparkmicro/Ki-nTree #36: Make the program work when your inventree instance has a self-signed cert

randrej, 2021-04-17:
The Python Inventree module, used here to communicate with Inventree, uses `requests` for communication. Requests by default use python-provided CA certificates, which means your self-signed cert won't work even if you installed it with `trust anchor <cert>.crt`. The solution is to export the env variable `REQUESTS_CA_BUNDLE=/etc/ssl/certs/ca-certificates.crt` before importing the inventree package, which forces it to use Linux system CA certs. This should be done on Linux only.

Do you agree? Shall I implement this?

eeintech, 2021-04-18:
Oh wow I didn't think certificates could make the `requests` package barf, sorry for that!

Your proposal looks good, happy to see it implemented 😃

randrej, 2021-07-05:
Yes, that's it. You'd have to access your Inventree instance through a reverse proxy that has SSL a self-signed cert. I'll provide a pull request.

PR #45 (2021-07-08): Use local CA certs for Inventree on Linux
This allows you to use your own, self-signed certificates on Linux.
If the '/etc/ssl/certs/ca-certificates.crt' file doesn't exist, the
program won't try to use local certificates.

None of this applies on non-Linux platforms.

Tested on my instance, works properly.

Fixes #36.
closed 2021-07-08 by eeintech