sparkmicro/Ki-nTree #37: Migrating to PySide2 or PyQT5?

randrej, 2021-04-18:
This may be an unpopular idea, and I'm fully aware why. I'm not even sure it does make sense at this point, but still, it's worth considering.

I saw that you considered migrating the project to PySimpleGUIQt. This does make sense, but that project is really in its infancy. Since the GUI part of this project isn't too complex, it would make sense to port the whole thing into raw PySide2.

### The benefits?
It's a much more polished library, with fewer bugs, and is a thin layer over Qt which is pretty solid. If you decide to add some more advanced functionality to this program, it's going to be much easier to do in PySide2/Qt than in a PySimpleGUI derivative. It's not too much work to port the app (because it's small), and it's not much harder to use generally, while allowing for improved functionality.

### Actual issues solved
Input field behavior on the current version (with PySimpleGUI) is abysmal: if you select  text and paste, the selected text doesn't get overwritten. Ctrl + A doesn't work. All this works beautifully in PySide2.

eeintech, 2021-04-19:
Hey @randrej I'm definitely open to improve the GUI, I do agree there are gaps for a better user experience. I only picked `PySimpleGUI` because I wanted a quick and easy way to make a very simple UI, just didn't want to spend much time on that part. I have not heard of `PySide2` but open to try it out!

eeintech, 2021-09-16:
I played a bit with `PySide2` but not convinced it brings much to the table... the Ki-nTree GUI is quite simple.

eeintech, 2022-01-13:
Closing for now. Can re-evaluate later if switch is necessary.

eeintech, 2023-01-06:
I stumbled this awesome looking & simple Tkinter GUI wrapper: https://github.com/TomSchimansky/CustomTkinter

Looks super nice and modern, even on Linux :smiley: 

[image]

42CrMo4, 2023-01-11:
Hello @eeintech, 
just wanted to mention two option you might want to take a look at. I do not have much experience with either of them but thought they might look interesting. 
Both have a calculator App as a reference project, so that what i picked as the screenshot ;) 

Textual is a TUI (Text User Interface) framework. The [homepage link](https://textual.textualize.io/) and [github repo link](https://github.com/Textualize/textual) .
Could be interesting because it is running in the terminal and has the ability to run remote over ssh.
[image]

and the Flet framework an python flutter alternative. The [homepage link](https://flet.dev/) and [github repo link](https://github.com/flet-dev/flet).
[image]

eeintech, 2023-01-11:
Thanks @42CrMo4 for mentioning those two frameworks.

I think Textual is not well fitted for general use, as it needs to be ran inside a terminal.

Flet looks really cool :+1:  I might give it a shot to compare it with CustomTkinter.

eeintech, 2023-01-13:
@42CrMo4 See below for Flet

[image]

[image]

[image]

[image]

42CrMo4, 2023-01-14:
@eeintech wow that was fast. looks good.

PR #122 (2023-02-21): 1.0.0-alpha1
## 1.0.0-alpha1 Release
- [x] Fixes #37: Moved GUI to [Flet](https://github.com/flet-dev/flet)
- [x] Fixes #87: Implemented DropdownWithSearch widget which lists the entire InvenTree category tree
- [x] Fixes #104: Support for InvenTree `0.9.x` and later, remove support for older versions
- [X] Fixes #119: Integrated KiUtils `v1.3.0` as PIP package

TODO for `alpha2`:
- [ ] #120 
- [ ] Implement feature parity with `0.6.x` (eg. InvenTree alternates)