sparkmicro/Ki-nTree #43: Expand Digi-Key Integration with more categories and ability to set InvenTree default options/field mapping?

Oakman-Dev, 2021-07-05:
Just two small suggestion that I can across when using the tool today (absolutely love it btw!)

- Would it be possible to expand the categories to cover things like thermal management and fans? Would love to see the tool pick up things like fan size, voltage, rpm and noise level from the Digi-Key listing.

- I would also love to see the option to choose to have the manufacturer's part number as default on the InvenTree Part Details for either the Part name or Description record, not just on Part Parameters (default options/field mapping).

eeintech, 2021-07-06:
Hello @Oakman-Dev, glad to know you like Ki-nTree!

> * Would it be possible to expand the categories to cover things like thermal management and fans? Would love to see the tool pick up things like fan size, voltage, rpm and noise level from the Digi-Key listing.

Documentation is lacking but here is what you need to know:
* categories and parameters are all configurable through `.yaml` files located in the `config` folder
* `config/inventree/categories.yaml` configures the category lists (first pop-up window)
* `config/digikey/digikey_categories.yaml` configures the mapping between the main categories defined above and Digi-Key's categories
    - Note that you can map a Digi-Key category to multiple main categories, using the `__` prefix on the category name (see `__Schottky` and `__Standard` subcategories under "Diodes")
    - If the `__` prefix is used, the "Function Type" parameter found on Digi-Key is used for category mapping (for the "Diodes" category the "Function Type" maps to "Diode Type" on Digi-Key, see `config/digikey/digikey_parameters.yaml` file)
 - The list of part parameters is stored in the `config/inventree/parameters.yaml`, if you update this file, you'll need to re-run the InvenTree setup script `setup_inventree.py` so that the new part parameters are created
 - Parameter mapping can be found in the `config/digikey/digikey_parameters.yaml`: it maps the InvenTree part parameters to Digi-Key's parameters/specification data

> * I would also love to see the option to choose to have the manufacturer's part number as default on the InvenTree Part Details for either the Part name or Description record, not just on Part Parameters (default options/field mapping).

That sounds doable, basically you would like something like follow?

[image]

W-Floyd, 2021-07-08:
The change of name to match the manufacturer would be a great help when working with connectors. For example:

[image]

There is nothing immediately clear that this is a JST XHP-2. In many other cases this is not an issue (I don't care which 10k resistor it is), but it is important at least for connectors.

eeintech, 2021-07-08:
@W-Floyd I would have to look to make the part data settings per category but that'd be more work than a universal rule/setting.

Replacing the data manually in the "Update Part Data" form, just for connectors, does not sound too bad though, no? Unless you have thousands of connectors that you want to add to your library...

W-Floyd, 2021-07-08:
Yeah, it'd be doable by hand in my case.

eeintech, 2021-07-09:
@W-Floyd Ok nice.

@Oakman-Dev What do you think of my interpretation for your feature request?

Oakman-Dev, 2021-07-11:
> @W-Floyd Ok nice.
> 
> @Oakman-Dev What do you think of my interpretation for your feature request?

I think you pretty much got it. I've "hacked" together with my own definition for the Fans, but as W-Floyd mentioned, it would be nice to see Fans and Connectors as standard parts in Ki-nTree. Especially since parts like MCUs, Fans and Connectors are more manufacturer and part-specific than a "generic" say a 10k 1% 0402 Resistor.

Oakman-Dev, 2021-07-11:
@eeintech Also, a small little bonus feature request. ;) The ability in settings to toggle if (after part creation) it opens up in the browser or not. Trying to enter hundreds of components with a cheap barcode scanner and your app works great. Just breaks workflow when I have to close the browser/switch back to your app.

eeintech, 2021-07-12:
@Oakman-Dev Ha yes sorry this was buried in the config files... 

Understood, I'll take a look at an implementation for more configurable part data schemes when I get the chance!

PR #51 (2021-07-28): Added user mapping for Digi-Key search keys
Fixes #43 

Mapping a search result key in `~/.config/kintree/user/digikey_config.yaml`:

[image]

Results in: 

[image]
closed 2021-07-28 by eeintech