sparkmicro/Ki-nTree #44: Digikey error clientid

Randomtyper23, 2021-07-07:
Hey all,

Are there any known problems with the linking of the program with the digikey-API? The installation succeeded on my PC, but when i enter the client and secret i get from following the video posted about the digikey API, i get an error (401) saying the clientID is invalid...
Does anyone know how to fix this?

Kind regards

eeintech, 2021-07-07:
@Randomtyper23 Just to be certain, have you followed the steps in this video to create a Digi-Key API instance: https://www.youtube.com/watch?v=OI1EGEc0Ju0

Are you sure it is a "**production**" app, not a sandbox?

Have you copied the Client ID and Key completely?

Randomtyper23, 2021-07-07:
Yes, i follow the video step by step :) 
It also display a production app yeah, and i copied and then manually typed it aswell once, but to no result :(

eeintech, 2021-07-07:
Ok good. Can you copy the error here or post a screenshot?

What browser are you using?

stijnloeys, 2021-07-07:
This is the error i get in chrome (also the browser i use): 
{"ErrorResponseVersion":"3.0.0.0","StatusCode": 401,"ErrorMessage":"Invalid clientId","ErrorDetails":"client_id query parameter has incorrect value. Send valid client_id","RequestId":"26583285-ef0b-41d2-c657-bf0594c1564e","ValidationErrors":[]}

the application says "failed to connect to digi-key Api" and the console says: 
2021-07-07 22:03:32,818 - digikey.oauth.oauth2 -  WARNING: Oauth2 token storage does not exist or malformed, creating new.
WARNING:digikey.oauth.oauth2:Oauth2 token storage does not exist or malformed, creating new.

Thanks in advance!

eeintech, 2021-07-07:
Oh maybe you have previous files that prevents the token to be formed correctly.

Check the path to your `Cache Folder` in the User settings:

[image]

then make sure this path does not have files (or delete them) other than two folders `images` and `search`.

Then re-try your Client ID and Key using Ki-nTree "Test" button.

stijnloeys, 2021-07-08:
Hey, I checked the cache folder and there were no files except for the images and search files you mentioned (they were both empty). But they are in a different folder than all the program itself:
[image]
while the program is in
[image]
with the following stuff inside:
[image]

eeintech, 2021-07-08:
@Randomtyper23 This is a really weird one...

Could you send a screenshot of your Digi-Key developer app settings (hiding the ID and Key)?
Maybe the callback URL is giving you some kind of issue.

StijnLoeys1999, 2021-07-08:
[image]
Did you mean this?

eeintech, 2021-07-08:
Yes, and it looks good indeed.

Do you see any file being generated in your cache folder when launching the Digi-Key API test from Ki-nTree?

StijnLoeys1999, 2021-07-08:
I re-installed it on my desktop and now it works! I truly don't know what I did different, but thanks!
Maybe a final question, how do i get the inventree adress? is that just the URL, because in your video it looks like an IP adress, but I don't have access to the computer it is running on

eeintech, 2021-07-08:
@StijnLoeys1999 Sorry I am confused, are you the same user as @Randomtyper23 and @stijnloeys ?

Yes it is just the URL, if you are running your InvenTree instance locally and did not setup a domain name, then it is just the IP and port in the URL, something like this should work: http://127.0.0.1:8000

StijnLoeys1999, 2021-07-08:
Yeah sorry, I switch computer a lot and everyone apparently has a different github account set-up, sorry for that...
The problem is that it is not setup locally :( the url that i use to access the database is "https://ivt.XXXXX.XXXXX/index/" but that does not work... (XXX's are to hide the site name) (dropping the https:// also does not work)

huge thanks for your help so far by the way! It really looks like an amazing program!

eeintech, 2021-07-08:
What if you drop the `index/` at the end of the URL?

StijnLoeys1999, 2021-07-08:
Also does not work then :/

The terminal says the following: "error connecting to InvenTree server: invalid address, username or password"

eeintech, 2021-07-08:
As you are on Windows, I don't think it is a certificate issue...

So with those exact same username and password, you can login to InvenTree at the "https://ivt.XXXXX.XXXXX/" URL?
I wonder if this is a SSL issue... I will investigate

StijnLoeys1999, 2021-07-08:
Yes I can log in, and thanks!

eeintech, 2021-07-08:
I don't think SSL is the issue, can you try to connect to my InvenTree demo instance?

server: `https://inventree.polarize.xyz/`
username: `visitor`
password: `showmethedemo`

StijnLoeys1999, 2021-07-08:
It worked

eeintech, 2021-07-08:
Do you know which port is the InvenTree instance running on?

I would suggest to try instead: "https://ivt.XXXXX.XXXXX:<PORT>" (replacing `<PORT>` with the actual port)

StijnLoeys1999, 2021-07-08:
I don't sorry :(, I tried all of the most common ports but to no succes (https://hostpapasupport.com/commonly-used-ports/)

eeintech, 2021-07-08:
@StijnLoeys1999 Sorry I don't really know why the API refuses to deliver the token for your URL :(
I'll try to google it later see if anything comes up

Randomtyper23, 2021-07-08:
Ok, no worries, thanks in advance

eeintech, 2021-07-08:
@Randomtyper23 Which server are you using?

For Apache, someone seemed to have a similar issue: https://stackoverflow.com/questions/66336931/django-drf-token-authentication-over-https-ssl

I would suggest to google the name of your server along with "drf api not working ssl"
You may encounter a setting on your server that would allow external requests.

eeintech, 2021-07-08:
Also, this seem to be outside of the scope of Ki-nTree, we have proven that your setup is up and running as you were able to connect to the demo version.

Feel free to update this ticket with your findings, however I won't be able to help much more, sorry :(

StijnLoeys1999, 2021-07-09:
No worries man, thanks for the help

eeintech, 2021-07-09:
Cool, I'm closing but feel free to post updates.

closed 2021-07-09 by eeintech