sparkmicro/Ki-nTree #47: Actions for pushing to PyPI

randrej, 2021-07-13:
We should set up Github Actions for pushing to PyPI. I propose we set up the actions in the following way:

1. The tests and style check are run
2. The package is pushed to Test PyPI
3. The package is pushed to actual PyPI

We need to decide on a source of truth for the package version. By default, Poetry expects the version to be in the `pyproject.toml` file. This can conflict with the version from the git tag.

In the near future, when `poetry 1.2.0` drops, we can switch to [poetry-version-plugin](https://github.com/tiangolo/poetry-version-plugin), which uses the version from the latest git tag.

Currently, we can do one of the following things:

### Automatically set the version from git tag

This is similar to the `poetry-version-plugin`. In the Github action that runs when something gets tagged, we use `sed` to replace the version variable in the `pyproject.toml` file before building. This guarantees that the version is always correct when we push the git tag. The downside is that, when you want to build the package locally, you need to manually specify the version. This package won't usually be built locally though, and every version on PyPI will be deployed by Github Actions, but it adds another step to local building. Who may want to build the package locally, anyway? A couple of us contributors, and maybe some Linux package maintainers. I think this is a sane trade-off.

### Check if the tag version and `pyproject.toml` version match

We can also expect that the version in `pyproject.toml` is always correct, and if it mismatches the tag version - an error is thrown and the package isn't pushed to PyPI. The downside to this is that if you, for example, have v0.4.13 in `pyproject.toml`, but you tagged the project with `v0.5.0`, the process fails and you can't fix the issue without force-pushing which is poor form. You'd have to post the fix as `v0.5.1` or do that force push.

I vote for the first option.

# PyPI and test PyPI

I think the easiest way to do this is if I set up everything on my fork, and test on 'Test PyPI', with my own tokens.

When I'm certain everything works properly, I'll transfer over the packages on PyPI and Test PyPI to you to maintain and own. You need to create accounts on PyPI and Test PyPI, and also to generate API tokens for pushing afterwards. If you plan on maintaining more packages, you should have one API key per package. You need to give me your PyPI username so that I can transfer the packages when I'm done.

eeintech, 2021-07-14:
Awesome plan, thanks @randrej

I like tagging so the first option to me makes sense. I'm even wondering if we can drop the version from the [version.yaml file](https://github.com/sparkmicro/Ki-nTree/blob/master/kintree/config/version.yaml) and use the poetry or tagged Git version in Ki-nTree directly?

As for the initial testing with PyPI it sounds great to me that you are willing to take the first stab.
Let me know if I can help with anything :smiley:

randrej, 2021-07-15:
I think #32 is finished, let's continue here.

randrej, 2021-07-15:
I have to figure out how to use my own secrets on my fork. This doesn't seem to work ATM, or I'm doing something wrong...

eeintech, 2021-07-15:
@randrej I fixed up some issues in #46 have you tried with those?

randrej, 2021-07-17:
Yes, I have, but to no success. I just figured out that you store the token in a git repo and that's why it's missing. I have to get the tests to work because the tests have to pass before I push to PyPI - you should only be able to push if the tests pass.

Can you give me all the secrets involved to copy over into my fork so that this just works and I can proceed? I won't abuse them and you can always invalidate them when I'm done and change them. You can send them to my e-mail, r.andrej@gmail.com.

eeintech, 2021-07-19:
@randrej I was going to but then I realize, what about if we skip the Digi-Key API test in the CI? We can always re-enable it later but I know right now it works fine, what do you think?

eeintech, 2021-07-19:
Never mind I just sent them to you!

eeintech, 2021-07-21:
@randrej Any luck with the CI tests? When do you think you can give PyPI a go?

randrej, 2021-07-21:
They now run OK on my fork, I'm currently configuring the PyPI deployment.

randrej, 2021-07-24:
My tests fail again for Digikey. I'll disable that part and re-enable them for the pull request.

randrej, 2021-07-24:
Digikey tests fail, but this should otherwise function. You need to tell me your Test PyPI username so that I can sign over the package to you. I haven't pushed anything to the regular PyPI, you have that honor.

Please alter the authors in `pyproject.toml`.

eeintech, 2021-07-24:
@randrej My username is the same as on GitHub: eeintech

I'll look into this soon, already saw some bits, can't wait to play with it! Nice job!

randrej, 2021-07-24:
You're now invited on `test.pypi.org`, with an owner role.

To be honest, this took way too much time. I *hate* Github Actions. Why didn't they just copy GitlabCI? It's so much easier to use and more ergonomic.

You'll have to fix Digikey tests.

closed 2021-07-26 by eeintech

eeintech, 2021-07-26:
I haven't used any of them, sorry it was a pain to work this out.

I think I did succeed in pushing to PyPI! On the first push, I realized the GUI version was still showing the older version, had to re-push again... but now it looks good!

`pip install kintree` now works!

Thanks for your great contribution :+1:

eeintech, 2021-07-26:
@randrej Any idea why the `README.md` is not pushed to PyPI? https://pypi.org/project/kintree/

PR #48 (2021-07-26): Actions for pushing to PyPI
When tags are pushed, tests and style checks are run, and then the package is built and published to Test PyPI and PyPI.

You need to add the following tokens (obtained from pypi.org and test.pypi.org) to the secrets:

```
TOKEN_TEST_PYPI
TOKEN_PYPI
```

I had to consolidate all the actions into one workflow, because that's the only way to avoid code duplication between on-push and on-tag actions.

Closes #47