sparkmicro/Ki-nTree #50: [Feature Request] Add ability items from Mouser and LCSC part numbers.

Oakman-Dev, 2021-07-26:
It would be great to expand the functionality to take Mouser and LCSC part numbers. 

I've thrown together a quick version (For LCSC) of this myself, modifying the digikey_search function and the snapeda example. It simply checks if the part_number begins with a 'C' and is between 5 and 7 digits letters long. Then it calls the https://wwwapi.lcsc.com/v1/products/detail?product_code={} endpoint to get the part data.

Mouser would be a bit trickier, but at least they have an official API at https://www.mouser.com/api-hub/

eeintech, 2021-07-26:
@Oakman-Dev I would love to be able to support more suppliers! It's quite a bit more work to do the categories and parameters mapping though...

How did you find out about LCSC api usage and where can I find doc about it? This URLs don't seem to work: https://github.com/yaqwsx/jlcparts/blob/master/LCSC-API.md

I haven't looked into Mouser's API yet, any python package to make it integration friendly?

Oakman-Dev, 2021-07-26:
@eeintech Funnily enough your current implementation seems to work quite well out of the box when applied to LCSC. I've been using it fairly successfully without even touching the mapping.

I think this is a new endpoint since they updated their website a few days ago. I found it myself whilst looking at their website traffic. It seems to just take the product_code like C344022 for example and spits out what's on the site. That also includes the link to datasheets and images. So I just remapped what you had already done for DigiKey.

PS: @eeintech Any chance you'd consider making a CLI/package version of Ki-nTree? Would love to be able to call your scripts from another function. Right now I'm manually copy-pasting from a spreadsheet generated by my code. In my case that would be an OpenCV barcode scanner that looks at the QR code on DigiKey, Mouser, LCSC bags, and adds the item and qty's automatic to InvenTree.

eeintech, 2021-07-27:
Very interesting! I'll look into LCSC soon then, sounds like a low-hanging fruit.
Or maybe you have a PR read for that? 😉 

I've never thought of a CLI for Ki-nTree, sounds also appealing but I _think_ it will require quite a lot of code refactoring to support this. I kinda bundled the GUI into one bigass main function and all functions are quite connected. Maybe what you are asking is the same bigass function but called from command-line instead of "button-triggered"?

Oakman-Dev, 2021-07-27:
> Very interesting! I'll look into LCSC soon then, sounds like a low-hanging fruit.
> Or maybe you have a PR ready for that? 😉
> 
> I've never thought of a CLI for Ki-nTree, sounds also appealing but I _think_ it will require quite a lot of code refactoring to support this. I kinda bundled the GUI into one bigass main function and all functions are quite connected. Maybe what you are asking is the same bigass function but called from command-line instead of "button-triggered"?

My Python skills are atrocious, but shoot me a message on Discord (or another platform of your choice), and I can send you the code that I have. I can also share the hacked together OpenCV/Webcam barcode reader that I built to use with your tool.

Re: CLI, yeah sort of. I was thinking (in an ideal world) that I could import the Ki-nTree package as kintree, and do something like kintree.add_part(part_no='C123456', supplier='LCSC', inventree=True, kicad=False).

eeintech, 2021-07-27:
Ha no worries, I'll see what I can do short term for LCSC, no promise on Mouser yet.

I may have some ideas for the CLI, will keep you posted.

eeintech, 2021-07-29:
@Oakman-Dev Wanna test the LCSC searches on the version I just released? `pip install -U kintree`

eeintech, 2021-08-16:
Made some progress on the Mouser API: https://github.com/sparkmicro/mouser-api

Hoping I can upload to PyPI soon and connect with Ki-nTree.

closed 2021-08-18 by eeintech

eeintech, 2021-08-18:
Added Mouser API support in [0.4.4](https://github.com/sparkmicro/Ki-nTree/releases/tag/0.4.4)

Oakman-Dev, 2021-09-05:
@eeintech Sorry for the delayed response, thank you so much for this! It seems to work great! Do I dare ask now, any progress on the CLI idea? ;)

eeintech, 2021-09-07:
@Oakman-Dev I had limited bandwidth in the last weeks, I thought about what you were trying to do and it would take a greater effort than I early anticipated. There is a flow to the GUI that could be reproduced via CLI, all I would do is wrap up all the steps into individual functions... which is quite useless for the sake of the GUI. So instead why not just calling Ki-nTree as a library in your python script and using the already existing functions? You can look at `main` function in the `kintree_gui.py` for a peek at the different steps and information flow.

PR #57 (2021-08-18): (0.4.4) Mouser API support
Fixes #50