sparkmicro/Ki-nTree #53: Check hard-coded version matches Git tag before posting to PyPI

eeintech, 2021-07-29:
This gets me every time...

randrej, 2021-07-30:
I can submit a PR that also changes that version in the git push action.

randrej, 2021-07-30:
Is this the right spot?

https://github.com/sparkmicro/Ki-nTree/blob/c0d1ac3de760abe108cbabaa8109e53ab3387e73/kintree/config/settings.py#L10-L14

eeintech, 2021-07-30:
@randrej Yes that's it, is it possible to edit the python code you think?

randrej, 2021-07-30:
Yes, sure. I can change these 3 lines in the code directly, or we could read them from another (python/json/yaml) file if you don't like mixing generated and written code.

eeintech, 2021-07-30:
@randrej It used to be in a YAML file but for the purpose of packaging I moved it to hard-coded inside the `settings.py`. I don't know what is the best at this point, is there maybe an easy way to hard-code a software version at the moment it is packaged using poetry?

randrej, 2021-07-30:
There are ways to obtain the version of a local package:

https://github.com/python-poetry/poetry/issues/273

To be honest, all of this just seems complicated and inefficient: let's just generate a `__version__.py` file when we update the `pyproject.toml`. This file contains that version dict and is imported in the `settings.py` file. For local development, you can provide your own `__version__.py` - the file isn't kept in git and is included in `.gitignore`.

eeintech, 2021-08-02:
@randrej I like this idea, do you want to implement it?

PR #58 (2021-08-24): Overwrite __version__ with tag value before publishing
Goal of this PR is to fix #53 

@randrej What do you think of this implementation?
closed 2021-08-24 by eeintech