sparkmicro/Ki-nTree #69: `0.4.7` won't start on Powershell/Windows

eeintech, 2021-10-29:
[image]

The `/` might be the problem!

deepthought-64, 2021-11-25:
This seems to be not only a window issue. I tried starting it today under Linux (Arch) and got the same error.
```
> python3 ./kintree
Traceback (most recent call last):
  File "/home/fabi/.local/bin/./kintree", line 8, in <module>
    sys.exit(kintree_gui.main())
  File "/home/fabi/.local/lib/python3.9/site-packages/kintree/kintree_gui.py", line 913, in main
    with open('images/logo.png', 'rb') as imagefile:
FileNotFoundError: [Errno 2] No such file or directory: 'images/logo.png'
```

eeintech, 2021-11-25:
Thanks for the report @deepthought-64 I hope to fix this issue soon!

closed 2021-11-26 by eeintech

eeintech, 2021-11-26:
@deepthought-64 Should be fixed in `0.4.8` :wink:

PR #71 (2021-11-26): 0.4.8
Maintenance release!

Fixes #70 (thanks to #64)
Fixes #69 
Fixes #67