sparkmicro/Ki-nTree #70: kintree_setup_inventree.exe not working on Windows system

Waranga-pa, 2021-10-31:
Hello and many thanks for creating this tool.

Setting up a new environment with Kicad and inventree, I was not able to initialize the inventree databases. Adding new parts with kintree fails with  
←[93m[INFO]     Warning: Parameter "Between Board Height" not found in supplier data←[0m
←[93m[INFO]     Warning: Parameter "Screw, Thread Size" not found in supplier data←[0m
←[93m[TREE]  Warning: Subcategory "Switch" does not exist←[0m
←[91m[ERROR]    Categories (Mechanicals - Switch) does not exist in InvenTree←[0m

As documentation is missing any hints I came to use  kintree_setup_inventree.exe to initialize inventree but it fails with

C:\Users\User\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_xxx\LocalCache\local-packages\Python39\Scripts> .\kintree_setup_inventree.exe
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.2032.0_x64__xxx\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.2032.0_x64__xxx\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\philipp\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\Scripts\kintree_setup_inventree.exe\__main__.py", line 7, in <module>
AttributeError: module 'kintree.setup_inventree' has no attribute 'main'

I used Python = 3.9 and pip = 21.3.1 for compiling.

Many thanks for any hints of what I'm doing wrong.

eeintech, 2021-11-17:
@Waranga-pa Please let me know about my previous questions :smiley:

deepthought-64, 2021-11-25:
Hey, if I want to start Ki-nTree under Linux (Arch) i get a similar problem when trying to invoke the setup:
```
> python ./kintree_setup_inventree
Traceback (most recent call last):
  File "/home/fabi/.local/bin/./kintree_setup_inventree", line 8, in <module>
    sys.exit(setup_inventree.main())
AttributeError: module 'kintree.setup_inventree' has no attribute 'main'
```

I have installed Ki-nTree via pip (`pip install -U kintree`). 
Ki-nTree version: 0.4.7
Python version: 3.9.7

eeintech, 2021-11-25:
@deepthought-64 Thanks I will run it the way you did and fix the issue.

deepthought-64, 2021-11-26:
Thanks a lot! I have cloned the git repo and invoked it manually and everything seems to work that way for me

closed 2021-11-26 by eeintech

eeintech, 2021-11-26:
@deepthought-64 @Waranga-pa Should be fixed in `0.4.8` :wink:

Waranga-pa, 2021-11-29:
Sorry for my late response, I only found the time to test right now and setting up Inventree database is working as expected now. Many thanks for your great work!

eeintech, 2021-11-29:
@Waranga-pa Awesome, great to know!

PR #71 (2021-11-26): 0.4.8
Maintenance release!

Fixes #70 (thanks to #64)
Fixes #69 
Fixes #67