sparkmicro/Ki-nTree #72: Symbol and footprint upload to KiCad fails

Waranga-pa, 2021-11-29:
I'm facing an issue I'm not 100% sure if it is related to my lack of understandig data transfer from Ki-nTree to Kicad, some misconfiguration or simply a bug. Error message looks like:

[MAIN] Adding part to KiCad←[0m
[KCAD] library_path: C:/Program Files/KiCad/share/kicad/library\Transistor_Array.lib
[KCAD] Number of parts in library Transistor_Array.lib: 20
←[91m[INFO] Error: Failed to add part to KiCad (incomplete InvenTree data)←[0m

My path variables are:
Symbol libraries path folder: C:\Program Files\KiCad\share\kicad\library
Footprint libraries folder: C:\Program Files\KiCad\share\kicad\modules
Symbol templates folder: C:\Users\user\AppData\Roaming\Python\Python39\site-packages\kintree\kicad\templates\

My kicad.yaml is untouched, not representing these changes set in GUI settings.

I tried with different part numbers and different suppliers and attached the complete console output:
Part: C20917		Supplier: LCSC
Part: C9667		Supplier: LCSC
880-1096-ND		Supplier: DigiKey
296-39101-1-ND		Supplier: DigiKey
493-13632-1-ND		Supplier: DigiKey

Many thanks

[debug_log.txt](https://github.com/sparkmicro/Ki-nTree/files/7618432/debug_log.txt)
s!

eeintech, 2021-11-29:
@Waranga-pa Thanks, I can't seem to be able to reproduce... When you get to the KiCad Libraries step, can you please send me a screenshot of the window before clicking on "Submit":

[image]

Waranga-pa, 2021-11-29:
[image]
[image]
[image]
[image]

eeintech, 2021-11-29:
@Waranga-pa That is rather weird it is not working, everything looks good in the GUI and debug log you sent...

Are you running Ki-nTree from pip or have you cloned this repository? We could alter the code and add some debug to see why is it failing on your system.
If you can find the `kintree/kintree_gui.py` file, on line 1253 (before the try statement) can you add: 
``` python
cprint(part_data)
```

And send the output of that debug print?

Thanks!

Waranga-pa, 2021-11-29:
I used the pip way before but now also tried the "venv"-way also. Even starting in an elevated cmd (admin-permission) didn't change anything. Here is the requested output, I don't see any mistakes:

C:\Users\user>kintree
←[95m
[MAIN]  Connecting to Inventree server←[0m
←[92m[TREE]     Successfully connected to InvenTree server (ENV=TESTING)←[0m

--------------------
←[95m
[MAIN]  Using LCSC cached data for C20917 ←[0m
←[95m
[MAIN]  Creating part in Inventree←[0m
[INFO]  Category: "Transistors"
[INFO]  Subcategory: "Load Switches"
[INFO]  User Category: "Transistors"
[INFO]  User Subcategory: "Load Switches"
←[93m[INFO]     Warning: Parameter "Vce Saturation (Max) @ Ib, Ic" not found in supplier data←[0m
←[93m[INFO]     Warning: Parameter "Vgs(th) (Max) @ Id" not found in supplier data←[0m
←[93m[INFO]     Warning: Parameter "Transistor Type" not found in supplier data←[0m
←[93m[INFO]     Warning: Parameter "FET Type" not found in supplier data←[0m
←[93m[INFO]     Warning: Parameter "Vgs (Max)" not found in supplier data←[0m
←[93m[INFO]     Warning: Parameter "Supplier Device Package" not found in supplier data←[0m
←[93m[INFO]     Warning: Parameter "Rds On (Max) @ Id, Vgs" not found in supplier data←[0m
←[93m[INFO]     Warning: Parameter "Current - Collector (Ic) (Max)" not found in supplier data←[0m
←[93m[INFO]     Warning: Parameter "Current - Continuous Drain (Id) @ 25°C" not found in supplier data←[0m
←[93m[INFO]     Warning: Parameter "Power - Max" not found in supplier data←[0m
←[93m[INFO]     Warning: Parameter "Power Dissipation (Max)" not found in supplier data←[0m
←[93m[INFO]     Warning: Parameter "Voltage - Collector Emitter Breakdown (Max)" not found in supplier data←[0m
←[93m[INFO]     Warning: Parameter "Drain to Source Voltage (Vdss)" not found in supplier data←[0m
←[93m[INFO]     Warning: Parameter "Operating Temperature" not found in supplier data←[0m
←[1m[INFO]      Part already exists, skipping.←[0m
←[95m
[MAIN]  Creating parameters←[0m
←[91m[TREE]     Error: Parameter template "Collector-Gate Voltage" does not exist←[0m
←[93m[TREE]     Warning: The following parameters were skipped:←[0m
--->    Value
--->    Collector-Gate Voltage
--->    Function Type
--->    Maximum Gate Voltage
--->    Package Type
--->    RDS On Resistance
--->    Rated Current
--->    Rated Power
--->    Rated Voltage
--->    Temperature Range
--->    Symbol
--->    Footprint
←[95m
[MAIN]  Creating manufacturer part←[0m
←[1m[INFO]      Manufacturer part already exists, skipping.←[0m
←[95m
[MAIN]  Creating supplier part←[0m
←[1m[INFO]      Supplier part already exists, skipping.←[0m
←[95m
[MAIN]  Adding part to KiCad←[0m
{
    "IPN": "TRA-000005-00",
    "category": [
        "Transistors",
        "Load Switches"
    ],
    "datasheet": "https://datasheet.lcsc.com/lcsc/1811081213_Alpha---Omega-Semicon-AO3400A_C20917.pdf",
    "description": "30V 5.7A 26.5m\u03a9@10V,5.7A 1.4W 1.5V@250\u03bcA N Channel SOT-23-3L MOSFETs ROHS",
    "image": "https://assets.lcsc.com/images/lcsc/900x900/20180914_Alpha---Omega-Semicon-AO3400A_C20917_front_10.jpg",
    "inventree_url": "http://192.168.40.18:1337/part/TRA-000005-00/",
    "keywords": "30V 5.7A 26.5m\u03a9@10V,5.7A 1.4W 1.5V@250\u03bcA N Channel SOT-23-3L MOSFETs ROHS",
    "manufacturer": {
        "Alpha & Omega Semicon": [
            "AO3400A"
        ]
    },
    "name": "AO3400A",
    "parameters": {
        "Collector-Gate Voltage": "-",
        "Footprint": "Package_TO_SOT_SMD:SOT-23",
        "Function Type": "-",
        "Maximum Gate Voltage": "-",
        "Package Type": "-",
        "RDS On Resistance": "-",
        "Rated Current": "-",
        "Rated Power": "-",
        "Rated Voltage": "-",
        "Symbol": "Transistor_Array:TRA-000005-00",
        "Temperature Range": "-",
        "Value": "AO3400A"
    },
    "revision": "A",
    "supplier": {
        "LCSC Electronics": [
            "C20917"
        ]
    },
    "supplier_link": ""
}
[KCAD]  library_path: C:\Program Files\KiCad\share\kicad\library\Transistor_Array.lib
[KCAD]  Number of parts in library Transistor_Array.lib: 20
←[91m[INFO]     Error: Failed to add part to KiCad (incomplete InvenTree data)←[0m
←[95m
[MAIN]  Opening URL http://192.168.40.18:1337/part/TRA-000005-00/ in browser←[0m

eeintech, 2022-01-13:
Hey @Waranga-pa Have you tried this with `0.5.0`? Do you still have the same issue adding part to KiCad V6 libraries?

eeintech, 2022-01-13:
I really cannot reproduce this error. 

As the part data looks good, it must be something the mechanism to write KiCad lib throwing out an error of some sort.
Maybe the Transistor_Array library is in read-only?

Going to close this one for now, feel free to re-open an issue if it persist with KiCad V6 and your own custom libraries (not the default ones in `C:\Program Files\KiCad\share\kicad\library`).

closed 2022-01-13 by eeintech