sparkmicro/Ki-nTree #74: Mouser import to InvenTree missing parameter data.

deepthought-64, 2021-11-29:
When I am importing data for a part from mouser into InvenTree, it seems as if Ki-nTree fetches Parameter data but it does not insert it (properly) into InvenTree. 

From the log it looks like it found (some of) the matching parameters and also populates them:
```
[MAIN]  Creating parameters
[INFO]  Success: The following parameters were created:
--->    Value
--->    Frequency
--->    Interface Type
--->    Locking
--->    Mating Height
--->    Mounting Type
--->    Number of Contacts
--->    Number of Rows
--->    Orientation
--->    Package Height
--->    Pitch
--->    Polarity
--->    Shielding
--->    Temperature Range
```
But when I look at the InvenTree page of the created part, all of them are empty (= having a dash '-' as value).
[image]

When I run the same exact part through the Digi-Key importer, (some of) the parameters get populated:
[image]

I Also get some `[INFO]  Warning: Failed to compare part with database` - I don't know if that is related or not.

Let me know if I can provide more info. 

I am running Ki-nTree from the GIT repo (most recent commit)
```
commit ef1dd4649c760b311bc58bd63687069be36df850 (HEAD -> main, tag: 0.4.8, origin/main, origin/HEAD)
Merge: ea93ccb 8fcd85c
Author: Francois <eeintech@eeinte.ch>
Date:   Fri Nov 26 15:35:37 2021 -0500
```

eeintech, 2021-11-29:
Hello @deepthought-64 I am guessing the output messaging is a little confusing... yes it did create all the parameters (based on the default list of category parameter templates) but those did not have any value found in the Mouser API response:

``` bash
[INFO]	Warning: Parameter "Frequency - Max" not found in supplier data
[INFO]	Warning: Parameter "Connector Type" not found in supplier data
[INFO]	Warning: Parameter "Flat Flex Type" not found in supplier data
[INFO]	Warning: Parameter "Locking Feature" not found in supplier data
[INFO]	Warning: Parameter "Fastening Type" not found in supplier data
[INFO]	Warning: Parameter "Mated Stacking Heights" not found in supplier data
[INFO]	Warning: Parameter "Mounting Type" not found in supplier data
[INFO]	Warning: Parameter "Number of Contacts" not found in supplier data
[INFO]	Warning: Parameter "Number of Positions" not found in supplier data
[INFO]	Warning: Parameter "Number of Rows" not found in supplier data
[INFO]	Warning: Parameter "Orientation" not found in supplier data
[INFO]	Warning: Parameter "Height Above Board" not found in supplier data
[INFO]	Warning: Parameter "Insulation Height" not found in supplier data
[INFO]	Warning: Parameter "Pitch" not found in supplier data
[INFO]	Warning: Parameter "Pitch - Mating" not found in supplier data
[INFO]	Warning: Parameter "Gender" not found in supplier data
[INFO]	Warning: Parameter "Shielding" not found in supplier data
[INFO]	Warning: Parameter "Operating Temperature" not found in supplier data
```

Looking on Mouser website, I thought it should have caught some of them: number of positions, rows, pitch, etc.

[image]

But unfortunately this is what the Mouser API responds:
``` json
{
    "Availability": "276 In Stock",
    "Category": "DIN 41612 Connectors",
    "DataSheetUrl": "https://www.mouser.com/datasheet/2/179/PDF_DS_09185407914_EN-2477163.pdf",
    "Description": "DIN 41612 Connectors 40P MALE STRT IDC SHORT LEVERS",
    "ImagePath": "https://www.mouser.com/images/harting/images/09185065902_SPL.jpg",
    "Manufacturer": "HARTING",
    "ManufacturerPartNumber": "09185407914",
    "MouserPartNumber": "617-09-18-540-7914",
    "PriceBreaks": [
        {
            "Currency": "CAD",
            "Price": "$5.91",
            "Quantity": 1
        },
        {
            "Currency": "CAD",
            "Price": "$4.97",
            "Quantity": 10
        },
        {
            "Currency": "CAD",
            "Price": "$4.72",
            "Quantity": 100
        },
        {
            "Currency": "CAD",
            "Price": "$4.13",
            "Quantity": 250
        }
    ],
    "ProductAttributes": [
        {
            "AttributeName": "Packaging",
            "AttributeValue": "Tray"
        },
        {
            "AttributeName": "Standard Pack Qty",
            "AttributeValue": "50"
        }
    ],
    "ProductDetailUrl": "https://www.mouser.ca/ProductDetail/HARTING/09185407914?qs=GZB%2FOSW0r0FZZKGaiCHzvg%3D%3D"
}
```

(tested also on their [API testing website](https://api.mouser.com/api/docs/ui/index#/SearchApi/SearchApi_SearchByPartNumber))

So all those parameters are missing indeed...

deepthought-64, 2021-11-29:
Oh, that is rather unfortunate. Does that mean that the Mouser API does not contain any "parameter" infos which can be used by Ki-nTree?

eeintech, 2021-11-29:
Unfortunately yes, the API is not responding with all data in their database... Which means nothing we can do other than filling data manually. Another workaround I sometimes use is to create the part using the Digi-Key API then add the mouser part as a new supplier part in InvenTree.

deepthought-64, 2021-11-29:
Ah cool, so when I have added the part with Mouser, can I also use Ki-nTree to add the Digi-Key parameters/supplier part?

deepthought-64, 2021-11-29:
Ah, sorry, I read your reply too fast. So is adding the part "twice" possible? Because if I remember correctly, it is not possible for Ki-nTree to check, if the part is already there (e.g. based on the MPN?)

eeintech, 2021-11-29:
@deepthought-64 It is not possible to match a "Mouser" part with a "Digi-Key" part right now because the comparison is done based on parameters. However I want (one day) to implement a option for adding a supplier part to an already existing InvenTree part, it is shown on the roadmap but not sure when I would have time for this... Maybe you do? :wink: 

For the time being the manual entry is required (to add the "Mouser" part as a supplier part) directly into the InvenTree UI. It is quite easy and relatively quick to do so :smiley:

deepthought-64, 2021-11-29:
Yeah, I understand the concept. A matching feature would indeed be super cool. 
But you don't want me to write that. I haven't touched Python since my days at the Uni and that is longer ago than I care to admit :wink:
Sorry I can't do more - But I'm there if you need feedback or ideas (I indeed have an idea about the whole Parameter functionality, but I am literally on my second (third if you count the day I installed it) day of using InvenTree and Ki-nTree so I want to get to know the tools a little bit better before I start suggesting features that might already exist)

deepthought-64, 2021-11-29:
I've closed this issue for now since this is not a Ki-nTree issue but apparently a shortcoming of the Mouser API.

closed 2021-11-29 by deepthought-64