sparkmicro/Ki-nTree #75: Whitespace handling in search string

markusEmm123, 2021-12-14:
**Description**
When i copy an article number from the mouser site and copy it to the search field, e.g.
"614-SMTU2032-LF", it copies the trailing \n with the article number.
This creates an error when searching.

**Error message**
Traceback (most recent call last):
  File "c:\users\username\appdata\local\programs\python\python37-32\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\username\appdata\local\programs\python\python37-32\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\Username\AppData\Local\Programs\Python\Python37-32\Scripts\kintree.exe\__main__.py", line 7, in <module>
  File "c:\users\username\appdata\local\programs\python\python37-32\lib\site-packages\kintree\kintree_gui.py", line 1046, in main
    part_supplier_info = inventree_interface.supplier_search(values['supplier'], values['part_number'])
  File "c:\users\username\appdata\local\programs\python\python37-32\lib\site-packages\kintree\database\inventree_interface.py", line 326, in supplier_search
    part_info = search_api.load_from_file(search_filename, test_mode)
  File "c:\users\username\appdata\local\programs\python\python37-32\lib\site-packages\kintree\search\search_api.py", line 14, in load_from_file
    part_data = config_interface.load_file(search_file)
  File "c:\users\username\appdata\local\programs\python\python37-32\lib\site-packages\kintree\config\config_interface.py", line 14, in load_file
    with open(file_path, 'r') as file:
OSError: [Errno 22] Invalid argument: 'C:/Users/username/[..somepath...]/KinTree/cache\\search\\Mouser_614-SMTU2032-LF\n.yaml'

**Expected Behaviour**
Trim whitespaces in search field.

eeintech, 2021-12-14:
Hello @markusEmm123 This is a rather weird one. So actually `\n` is a line feed so I wonder if you are only copying the part of the Mouser part number and not data underneath? It can be tricky sometimes!

markusEmm123, 2021-12-15:
Yes, it's pretty much possible that Chrome does strange things when i select not exactly to the end of the line. Or maybe it catches ne newline-character when selecting the line up to the end.

eeintech, 2021-12-15:
@markusEmm123 Yes maybe it is the case. I'm not really sure I want to clean up the input part number for all copy-paste manipulation ambiguity. Were you able to try a different copy-paste technique to make it work?

markusEmm123, 2021-12-15:
The easy solution was deleting the whitespace(s) manually from the input field.
Do you read the input as string? Can the String.strip()-function handle this with little effort?

eeintech, 2021-12-16:
Yes input part number is treated as a string, however I'm reluctant of the idea of using any post processing function on the input string to avoid altering it.

markusEmm123, 2021-12-16:
Thank you for the information.
In my opinion you could strip whitespaces at the end and beginning of the string.
But this is just my opinion :)

I won't complain about an otherwise really great tool, please keep up the good work!

closed 2021-12-16 by markusEmm123

eeintech, 2021-12-17:
Thanks, let's keep an eye on this if it comes up often we can get that proposed fix in