sparkmicro/Ki-nTree #79: Error connecting to InvenTree server: invalid address, username or password

advlogic, 2022-02-06:
Cannot connect to InventTree server, despite being able to login via web with same credentials via web browser.  IventTree running successfully on Linux otherwise.  Digikey API works fine.

InventTree address:  192.168.0.213:1337
Username: xxx
Password: xxx matches InventTree logins

eeintech, 2022-02-07:
Hello @advlogic I am not able to reproduce this issue, I've tried on multiple versions.

What does this command give you?

``` bash
pip freeze | grep inventree
```

Try to erase the password completely and re-type it/paste it one more time.

advlogic, 2022-02-08:
No go on the password re-type.  It is 100% correctly set.

Output of pip freeze | grep inventree:

/usr/lib/python3.10/site-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 2.0.5-build-libtorrent-rasterbar-src-libtorrent-rasterbar-2.0.5-bindings-python is an invalid version and will not be supported in a future release
  warnings.warn(
inventree==0.4.4

eeintech, 2022-02-08:
Seems like you are using python 3.10? I haven't tested Ki-nTree with this version, currently the version supported are 3.7 to 3.9.

advlogic, 2022-02-09:
That looks like it may be the issue.  Perhaps I will revisit in the future should 3.10 be supported.  Thanks.

closed 2022-02-09 by advlogic

eeintech, 2022-02-09:
It might not be it... maybe something else with your setup. Have you tried another python version?

a7hybnj2, 2024-10-10:
@va7wv did you ever get this working? I installed kintree in venv with py3.8 and I get this same error.

```
kintree
2024-10-10 10:55:18.060 Python[45710:22361925] +[IMKClient subclass]: chose IMKClient_Legacy
2024-10-10 10:55:18.060 Python[45710:22361925] +[IMKInputSession subclass]: chose IMKInputSession_Legacy
[TREE]	Error connecting to InvenTree server: invalid address, username or password
[TREE]	Error connecting to InvenTree server: invalid address, username or password
```

despite the address, user, pass being correct.

va7wv, 2024-10-13:
No, I have switched my workflow to Part-DB.  I had too many issues trying to get IvenTree going in general.

> @va7wv did you ever get this working? I installed kintree in venv with py3.8 and I get this same error.
>
> ```
> kintree
> 2024-10-10 10:55:18.060 Python[45710:22361925] +[IMKClient subclass]: chose [...]

a7hybnj2, 2024-10-13:
@va7wv Thanks for getting back to me. Bummer. I spun up a "inventree production" instance thinking it might be dev vs production but I have the same error ¯\\_(ツ)_/¯.

va7wv, 2024-10-15:
I spent a lot of time on Iventree as well but it was just too much to try and get working.  Luckily it was in a VM.  For my purposes, Part-DB has worked out really great.  If you are keeping track of electronics parts, I highly recommend it.

> @va7wv Thanks for getting back to me. Bummer. I spun up a "inventree production" instance thinking it might be dev vs production but I have the same error ¯\_(ツ)_/¯.