sparkmicro/Ki-nTree #82: Documentation Addition

kontakt, 2022-03-14:
The minimum permissions required for a parts-creation user are as shown:
[image]

I'm not sure if the View/Change perms are required.
It is good practice to limit automated tools to the bare minimum permission set they need, and I think it would be ideal to have these defined in documentation.

eeintech, 2022-03-14:
Thanks @kontakt

You are referring to InvenTree user permission while this is the Ki-nTree repository :smiley: 
I would suggest to post here for a proposed improvement to the documentation for third-party tools permissions:
https://github.com/inventree/inventree-docs/issues

kontakt, 2022-03-16:
That is one way to approach that, but it can also be approached by showing which permissions Ki-nTree requires to function.

eeintech, 2022-03-16:
Well good point, I could have a note about this :smiley:

PR #84 (2022-03-25): 0.5.2 Release
- [x] Fix LCSC API data download - fixes #83
- [x] Add setting to automatically create missing sub-category - fixes #86 
- [x] Add documentation for starting a new InvenTree setup from scratch - fixes #80
- [x] Add documentation to InvenTree permissions - fixes #82
closed 2022-03-25 by eeintech