sparkmicro/Ki-nTree #85: [Feature Request] Ability to call Ki-nTree from other Python scrips to create a part (without GUI)

Oakman-Dev, 2022-03-22:
It has been discussed before, but I'll open up this feature request on its own. It would be great to have the ability to call Ki-nTree from another python script, bypassing the GUI and feeding it parameters like: kintree.add_part(part_no='C123456', supplier='LCSC', inventree=True, kicad=False)

This would be ideal when loading/creating a bunch of components from a database or API.

eeintech, 2022-03-22:
Hello @Oakman-Dev The problem relies in selecting things: confirming part data, categories, KiCad template and footprint... We would have to assume all of them to use found defaults.
Removing the KiCad constraint might reduce the effort, but again InvenTree categories might not be found automatically and the method wouldn't know what to do next...

eeintech, 2022-04-11:
Hi @Oakman-Dev 

Are you still interested to dig into a command-line tool?

eeintech, 2022-05-02:
No response from OP, closing.

closed 2022-05-02 by eeintech