sparkmicro/Ki-nTree #86: [Feature Request] Automatically create missing sub-category

Oakman-Dev, 2022-03-22:
To ease organising at component creation in Inventree, it would be beneficial to have Ki-nTree automatically create a new sub-category (or category even) if it's not found in Inventree. This would be useful for times when the existing categories does not match, and one has not been created in Inventree yet.

eeintech, 2022-03-25:
@Oakman-Dev Please try the new release (`0.5.2`). You can enable the setting in your `~/.config/kintree/user/general.yaml` file:

```
AUTOMATIC_SUBCATEGORY_CREATE: true
```

closed 2022-03-25 by eeintech

Oakman-Dev, 2022-03-26:
@eeintech This works flawlessly, thank you! 

> [TREE]	Warning: Subcategory "Wire-To-Board" does not exist
[TREE]	Success: Subcategory "Connectors/Wire-To-Board" was automatically added to InvenTree

PR #84 (2022-03-25): 0.5.2 Release
- [x] Fix LCSC API data download - fixes #83
- [x] Add setting to automatically create missing sub-category - fixes #86 
- [x] Add documentation for starting a new InvenTree setup from scratch - fixes #80
- [x] Add documentation to InvenTree permissions - fixes #82