sparkmicro/Ki-nTree #87: Support hierarchical categories

exp, 2022-04-05:
Ki-nTree is designed with the idea of a top level parent category, and then a subcategory. This is a little restrictive and doesn't match how my categories are set up in Inventree.

I've prepared a patch to implement hierarchical categories for the setup function and the user category select screen, but before proposing a PR I thought I'd open an Issue for discussion.

I don't understand why Digikey's categories are merged with the user's category list in this screen, nor the rationale behind integration of categories with Kicad. I think there are a few valid options for how these could be handled, so if you are open to adding this feature could you let me know how you'd like it integrated best?

Thanks.

eeintech, 2022-04-11:
Hello @exp 

Did you have more questions regarding the use of categories in Ki-nTree?

exp, 2022-04-11:
Hi @eeintech no that's fine, I am just trying to find time to get this coded. I'm having troubles receiving from Digikey and I would like to add a search results box too. I'll break the commits up appropriately.

I should have time to work on it this week, but I have to finish the actual PCBs first ☺

eeintech, 2022-04-11:
Ok sounds good, good luck with the PCBs!

eeintech, 2022-05-02:
Hi @exp How are the PCBs doing? Are you still planning to support this issue with a PR?

exp, 2022-05-02:
Hi @eeintech I will get to this issue as soon as possible. I've had issues with getting any results at all from Digikey which makes it quite difficult to test and I've had to jump to other projects. Sorry for the delay.

eeintech, 2022-05-02:
No problem, thanks for the update!

eeintech, 2022-06-30:
@exp Any progress? :smile:

martonmiklos, 2022-07-21:
Hi folks,

Once upon a time I was thinking about making a 'supplier API' -> Inventree conversion program, but luckily I found Kin-tree before starting the actual coding.

My original idea was to provide an Inventree category tree selector to the user in the part add dialog.

When a part is added to a specific category in Inventree from a supplier category then the match would be saved to a database (SQLite for e.g.). If a category match found in this db for a new part the Inventree category would have been already selected in the new part dialog.
Over time this would have greatly reduced the category selection task.

eeintech, 2022-07-21:
@martonmiklos I have implemented some kind of automatic matching process in the category selection, either by:
- matching to supplier categories
- remembering the user choice on the last supplier category match

Check out the `get_categories` method:
https://github.com/sparkmicro/Ki-nTree/blob/253f248e17d36dd9a989fd97318410474e4d5182/kintree/database/inventree_interface.py#L45

eeintech, 2022-07-22:
@exp What implementation scheme did you have in mind? How would you improve the categories window?

[image]

Weissnix4711, 2022-07-24:
Just my two cents, but how about copying the category dialog from Inventree?

[image]

Instead of two seperate dropdowns for category and subcategory, have one dropdown which lists all available categories. Only problem is the custom category name option. You could again copy Inventree, and have a dedicated 'Add category' form, or just let the user type their own full category path. Eg, `Capacitors/Aluminium/SMD`. You'd need to parse that somehow, figure out which (sub)categories already exist, and create the subcategories as necessary.

exp, 2022-07-25:
Hi All,
  I apologise for my tardiness on this issue. My PCB project is still a high priority and I took some time off inbetween. This week I am free to work on these projects. I want to get a patch into Inventree but then I am fine with any approach.

  @eeintech I planned to make the application a single unified window where possible, with a drop-down as Weissnix suggests. My inventree has quite extensive categorisation so I want to make sure they're fully supported as deep as possible, it's the matching of supplier categories I need to take a bit of time to do.

Sorry again for the delay,
Cheers!

eeintech, 2022-07-25:
> Just my two cents, but how about copying the category dialog from Inventree?
> 
> [image]
> 
> Instead of two seperate dropdowns for category and subcategory, have one dropdown which lists all available categories, subcategories, sub-subcategories, etc. Only problem is the custom category name option. You could again copy Inventree, and have a dedicated 'Add category' form, or just let the user type their own full category path. Eg, `Capacitors/Aluminium/SMD`. You'd need to parse that somehow, figure out which (sub)categories already exist, and create the subcategories as necessary.

Ha, I like this idea! So simple and efficient, no reinventing the wheel.

@exp Let me know where can I can give you a hand, I would like to get this implemented soon :smile:

exp, 2022-07-26:
@eeintech I ideally need to address the Inventree issue I opened regarding part boxes and storage first but I appreciate I've delayed you plenty on this. If you want to take the lead I'd really appreciate that but if not I'll try and contribute some examples in the next day.

eeintech, 2022-07-27:
@exp I picked up some workload recently and have to prioritize this, as soon as I am done I wish to look into this, if you haven't already :smiley:

Fivefold, 2022-08-17:
It's probably obvious to @eeintech but I just want to note that for proper hierarchical category support the structure of both [categories.yaml](https://github.com/sparkmicro/Ki-nTree/blob/84be9ddc8bd5027b9dffaa7a208ef63092d20b3e/kintree/config/inventree/categories.yaml) and [supplier_parameters.yaml](https://github.com/sparkmicro/Ki-nTree/blob/84be9ddc8bd5027b9dffaa7a208ef63092d20b3e/kintree/config/search/supplier_parameters.yaml) would need some changes. 

1. Currently (if my interpretation is correct, please correct me if I'm wrong) the "codes" (the abbreviations for the inventree-internal part number, e.g. TRA for transistors) only support top-level categories.
https://github.com/sparkmicro/Ki-nTree/blob/84be9ddc8bd5027b9dffaa7a208ef63092d20b3e/kintree/config/inventree/categories.yaml#L2-L18
A solution would be to match the keys in the code section with categories, regardless of nesting level. Of course this would prevent two different categories with the same name to have different codes, but that seems like an unlikely use-case.

2. Categories would also need an overhaul
https://github.com/sparkmicro/Ki-nTree/blob/84be9ddc8bd5027b9dffaa7a208ef63092d20b3e/kintree/config/inventree/categories.yaml#L20-L23 \
Here the easiest change I could think of would be something like this: 
```yaml
CATEGORIES:
  Passive Components: # Level 0
    Capacitors:       # Level 1
      Ceramic:        # Level 2
      Electrolytic:   # Level 2
    Inductors:        # Level 1
    Resistors:        # Level 1
```
Of course this mixes values and keys depending on context so there are probably better solutions. I'm no yaml guru unfortunately.

3. In supplier_parameters.yaml there doesn't seem to be support for subcategories even now. Direct name matching like in 1. would probably be the easiest way (for the user).

eeintech, 2022-08-18:
@Fivefold There is indeed a little bit of thinking for deeper category trees.

> 1. Currently (if my interpretation is correct, please correct me if I'm wrong) the "codes" (the abbreviations for the inventree-internal part number, e.g. TRA for transistors) only support top-level categories.

That's correct and if user picks a sub-category, the code used is from its parent.

> 2. Categories would also need an overhaul

Right, it could be updated to a dictionary based on InvenTree categories.

> 3. In supplier_parameters.yaml there doesn't seem to be support for subcategories

Indeed to lower the complexity, parameters are shared across top-level categories, and not sub-categories.

Here are some thoughts, dreaming a tiny bit bigger:
- Category tree fetched from InvenTree directly, with an expiration date to avoid fetching every single search
- Auto-update to `categories.yaml` file, move codes to a separate one
- In category selection window: propose existing InvenTree categories in InvenTree format (see @Weissnix4711 proposal above)
- Give user option to enter a new category (eg. `ICs/Logic/And Gate`) and create it on the fly, if user rights permit
- Auto-fetch and auto-update of `supplier_parameters.yaml` based on InvenTree categories parameters, with an expiration date to avoid fetching every single search
- Extra GUI step/window showing a table containing: supplier parameter name, supplier parameter value, dropdown with exisiting InvenTree categoy parameters and extra field for creating a new category parameter. If a match was already found, use it as default value inside dropdown box. If extra field is filled in: create parameter in InvenTree and ignore dropdown value.

martonmiklos, 2022-10-31:
I started to work on this, I have a dynamically populated (from InvenTree) treeview:
[image]

I plan to add a recent list next to it with the latest imported categories.

eeintech, 2022-10-31:
@martonmiklos Nice work! Any chance you could share your implementation with us?

martonmiklos, 2022-10-31:
@eeintech The code lives here: 
https://github.com/martonmiklos/Ki-nTree/tree/rething_category_selection

It is above a TME support branch, but I plan to rebase to master when I am done.

eeintech, 2022-11-01:
@martonmiklos Thanks, I'll try to take a look soon :smiley:

martonmiklos, 2022-11-02:
> @martonmiklos Thanks, I'll try to take a look soon smiley

Okay just for sync up here is my vision on how should it look like (in the case if InvenTree export is selected):

Ki-n-tree has an InvenTree PK - supplier category identifier lookup
User enters the partnumber, data queried from the supplier
If the supplier category has match in the lookup then in the second page under here:
[image]
We should write the category path from InvenTree plus an Edit button.
In this case if the user hit the submit button the part shall be created without any further dialogs.

In the case if no mapping found this line shall be hidden and when the user hits the submit the category selection dialog shall be shown next.
[image]
If there is an existing mapping in the recent list this shall be the topmost item, and shall be selected by default.
The same dialog shall be opened in the case if the user selects the Edit button next to the Category path.

eeintech, 2022-11-09:
@martonmiklos I'm thinking to first merge and cut the `0.6.0` release soon as it brings a bunch of new stuff. Then work on the category revamp. What do you think?

martonmiklos, 2022-11-10:
The improved category selection is heavily WIP, and I have no idea when will I find some time to finish. So my recommendation is to go ahead, I will handle the rebase on my fork.

martonmiklos, 2022-11-10:
I apologize I forgot to reply to this one.

> If I'm not mistaken, that would be two ways to select categories with two different flows (first is nested inside the part form window and second is **after** the part form window, as it is now). I'm not sure why you want those two flows... I would prefer to stick to your original idea and, when there is no match, using the edit button would be the only way to select the category. The validation of the form, instead of being currently done only on "Name" and "Description", would also be done on the category selection. I think it would simplify the flow so there is only one entry point for the category selection. What do you think?
> 

You might misunderstood me. My plan is to make a single window to select the category, and it would allow two way for category selection: tree based approach from InvenTree and selection from the recent list.

The current flow is :
Enter supplier PN -> Edit supplier data -> select category -> create part

In the case if there is category match it is reduced to:
Enter supplier PN -> Edit supplier data -> create part

If there is no category match then the current flow is being executed.

> The Edit category window you showed looks cool, is it functional? If it is, I have some time on my hands right now, I could finish the implementation if you'd like. We could merge you branch into `0.6.x` and it would add your changes to the current opened PR.

The tree selection side is functional the recent list is not yet. From my side this week is quite busy, if you can move things forward on my work feel free to do so. I can catch up

eeintech, 2023-01-05:
Hello @martonmiklos @exp 

I have gotten feedback on the category selector and have found multiple issue with old code which I want to fix now and release in a newer version of Ki-nTree (thinking `0.7`).
That's why I would like to go ahead on this implementation you started @martonmiklos in this branch?
https://github.com/martonmiklos/Ki-nTree/tree/rething_category_selection

martonmiklos, 2023-01-05:
Hey,

Uhm yes. I messed up that branch a bit as I used to...
It contains 3 commits from the master.
- Remember last supplier -> this had been merged
- Support for TME import. I wanted to PR it, but I wanted to have some discussion on this before
- Aand the the category selector changes in the question

Before moving forward I need to separate these things, preferably rebase the TME and merge that first.

eeintech, 2023-01-05:
@martonmiklos What is TME? Happy to discuss it :+1: 

If you do some clean-up let me know!

eeintech, 2023-02-16:
FYI I have implemented it in the new GUI, target is `1.0.0` release:

[image]

hvraven, 2023-03-16:
Hi @eeintech ,
thanks for implementing this. I am currently testing the new 1.0.0b1 release and it looks really nice. I am however not able to configure the `parameters_filters.yaml` and supplier categories (`digikey_categories.yaml`) correctly to make them work with hierachical categories.  We store other components in Inventree and thus have all electronics below a top level `Electronics` category. I tried nesting the categories in the yaml, as well as using forward slashes in the name directly.
Is this supposed to be working already? If you could give me a pointer on where to start looking I could also try to help with the implementation.

eeintech, 2023-03-17:
Hello @hvraven 

Actually I was thinking to deprecate the Supplier category lookups for `1.0.0` as it is quite easier to select a category now...

The `parameters_filters.yaml` file works for me, it will only take effect with `Digi-Key` searches at the moment as other suppliers parameter mapping is missing. Do you have the problem with `Digi-Key`? Can you please show me the output of Ki-nTree in your terminal?

Also do you mind sharing a overall glimpse at your category mapping (you don't have to share the full file, just let me know the structure so I can try to reproduce)?

Thanks!

PR #122 (2023-02-21): 1.0.0-alpha1
## 1.0.0-alpha1 Release
- [x] Fixes #37: Moved GUI to [Flet](https://github.com/flet-dev/flet)
- [x] Fixes #87: Implemented DropdownWithSearch widget which lists the entire InvenTree category tree
- [x] Fixes #104: Support for InvenTree `0.9.x` and later, remove support for older versions
- [X] Fixes #119: Integrated KiUtils `v1.3.0` as PIP package

TODO for `alpha2`:
- [ ] #120 
- [ ] Implement feature parity with `0.6.x` (eg. InvenTree alternates)