sparkmicro/Ki-nTree #88: Bulk importing to inventree

spectre-ns, 2022-05-06:
I'm trying to bulk import parts from digikey into inventree is this possible with your tool?

eeintech, 2022-05-09:
Hello @spectre-ns 

Yes with a little bit of scripting this is totally doable!

Take a look at how the CI tests are ran: https://github.com/sparkmicro/Ki-nTree/blob/main/run_tests.py
It's taking a list of parts located in this file: https://github.com/sparkmicro/Ki-nTree/blob/main/tests/test_samples.yaml
And adding them one by one in InvenTree (and KiCad libs). 

In the CI tests I'm actually cheating as I'm using the previously fetched data on all those parts, but you can make it so it has to fetch each part's data from Digi-Key instead.

Hopefully this gets you started, let me know if you have any question!

spectre-ns, 2022-06-14:
Answered! Thanks!

closed 2022-06-14 by spectre-ns

Dylancyclone, 2022-06-18:
In case anyone else would like a quick and dirty script for bulk importing, I found myself needing the same thing and quickly whipped one up [here](https://github.com/Dylancyclone/Ki-nTree/commit/a9c2ee7072b21af11db13fcf31c4644444c3159b). Much like the testing script, it can be run with `poetry run python -m bulk_import` and simply pulls from a `bulk_import.yaml` file next to the script.

eeintech, 2022-06-20:
@Dylancyclone That's cool, thanks for sharing!