sparkmicro/Ki-nTree #90: AttributeError: 'NoneType' object has no attribute 'pk'

KenwoodFox, 2022-06-08:
Installed in pipenv because the aur package depends on packages i cant seem to find, `python-inventree`?

Full traceback:

```
❯ kintree

[MAIN]  Connecting to Inventree server
[TREE]  Successfully connected to InvenTree server (ENV=TESTING)

--------------------

[MAIN]  Using Digi-Key cached data for 2616-VZH100M1CTR-0406S-ND

[MAIN]  Creating part in Inventree
[INFO]  Category: "Capacitors"
[INFO]  Subcategory: "Aluminium"
[INFO]  User Category: "Capacitors"
[INFO]  User Subcategory: "Aluminium"
[INFO]  Warning: Parameter "Temperature Coefficient" not found in supplier data
Traceback (most recent call last):
  File "/home/joe/.local/share/virtualenvs/joe-xljV_iqz/bin/kintree", line 8, in <module>
    sys.exit(main())
  File "/home/joe/.local/share/virtualenvs/joe-xljV_iqz/lib/python3.8/site-packages/kintree/kintree_gui.py", line 1164, in main
    new_part, part_pk, part_data = inventree_interface.inventree_create(part_info=part_info,
  File "/home/joe/.local/share/virtualenvs/joe-xljV_iqz/lib/python3.8/site-packages/kintree/database/inventree_interface.py", line 372, in inventree_create
    subcategory_pk = inventree_api.get_inventree_category_id(category_name=subcategory_name,
  File "/home/joe/.local/share/virtualenvs/joe-xljV_iqz/lib/python3.8/site-packages/kintree/database/inventree_api.py", line 53, in get_inventree_category_id
    cprint(f'[TREE]\t{item.getParentCategory().pk} ?= {parent_category_id}', silent=settings.HIDE_DEBUG)
AttributeError: 'NoneType' object has no attribute 'pk'
```

Not exactly sure if its an issue with just this part or others, thank you for making this software though! this should automate a lot!

KenwoodFox, 2022-06-08:
I added a `if item.getParentCategory() is not None:` at line 49 in the above file, that got me here

[image]

Witch is odd... because

[image]

KenwoodFox, 2022-06-08:
[image]

Nope, still no go, not sure what ive done wrong

KenwoodFox, 2022-06-08:
[image]
[image]

Its not the subcatagory thats missing... its the main catagory,

KenwoodFox, 2022-06-08:
So, what ive found is `inventree_api.get_inventree_category_id("String")` returns the correct id for any subcategory i query. But it always returns `-1` for a top level category, witch is odd because the way it looks to be setup, u need to return the id of a parent category, before querrying for a child one...

KenwoodFox, 2022-06-08:
Woah, omg i got it working, this may be dumb but, heres what i did

```diff
diff --git a/kintree/database/inventree_api.py b/kintree/database/inventree_api.py
index 3b8517a..2ab3a7e 100644
--- a/kintree/database/inventree_api.py
+++ b/kintree/database/inventree_api.py
@@ -46,7 +46,7 @@ def get_inventree_category_id(category_name: str, parent_category_id=None) -> in
     part_categories = PartCategory.list(inventree_api)
 
     for item in part_categories:
-        if item.getParentCategory() is not None:
+        if True:
             if category_name == item.name:
                 # Check parent id match (if passed as argument)
                 match = True
```

its more of just a patch but when it dosent check for the parent being none, it can return the right id for parent and child and now it works.

KenwoodFox, 2022-06-09:
@eeintech Im using 0.5.3

```
❯ pip freeze
...
digikey-api==0.6.1
inventree==0.7.1
kintree==0.5.3
mouser==0.1.1
```

eeintech, 2022-06-09:
Well... no idea where that line came from 😅 Could be leftover from a much older version?

KenwoodFox, 2022-06-09:
I couldent tell you :c i tried updating but python says 0.5.3 is the latest, so, idk, ive patched my version with my diff above, odd?

closed 2022-06-09 by KenwoodFox