sparkmicro/Ki-nTree #96: ImportError: cannot import name 'Iterable' from 'collections' (possible python 3.10 issues on aur package?)

KenwoodFox, 2022-07-26:
When i run kintree i get the following:

```
❯ kintree
Traceback (most recent call last):
  File "/usr/bin/kintree", line 33, in <module>
    sys.exit(load_entry_point('kintree==0.5.3', 'console_scripts', 'kintree')())
  File "/usr/bin/kintree", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/usr/lib/python3.10/site-packages/kintree/kintree_gui.py", line 15, in <module>
    from .search import digikey_api as digikey_api
  File "/usr/lib/python3.10/site-packages/kintree/search/digikey_api.py", line 3, in <module>
    import digikey
  File "/usr/lib/python3.10/site-packages/digikey/__init__.py", line 2, in <module>
    from digikey.v2.api import (search, part)
  File "/usr/lib/python3.10/site-packages/digikey/v2/api.py", line 7, in <module>
    from digikey.v2 import models
  File "/usr/lib/python3.10/site-packages/digikey/v2/models.py", line 8, in <module>
    from schematics.exceptions import ConversionError, DataError, ValidationError
  File "/usr/lib/python3.10/site-packages/schematics/__init__.py", line 6, in <module>
    from . import deprecated
  File "/usr/lib/python3.10/site-packages/schematics/deprecated.py", line 8, in <module>
    from .types.serializable import Serializable
  File "/usr/lib/python3.10/site-packages/schematics/types/__init__.py", line 2, in <module>
    from .base import *
  File "/usr/lib/python3.10/site-packages/schematics/types/base.py", line 19, in <module>
    from collections import Iterable, OrderedDict
ImportError: cannot import name 'Iterable' from 'collections' (/usr/lib/python3.10/collections/__init__.py)
```

Python 3.10 migration issues?

```
❯ pacman -Q python-kintree
python-kintree 0.5.3-1
```

Let me know if theres anything more i can provide!

eeintech, 2022-07-27:
Hello @KenwoodFox, thanks for reporting this issue.

Regarding the root cause, I have requested the v2 Digi-Key API to be removed as it is the faulty source and has been deprecated: https://github.com/peeter123/digikey-api/issues/26#issuecomment-1197353950

You could try to switch to Python `3.9` and re-try, it should work.

It is interesting that the Github action did not fail on this :thinking:

KenwoodFox, 2022-07-28:
Hope it gets updated soon, i did try getting kintree working in a virtualenv but, might just wait to get the aur package working with 3.10 again. Thanks for the info

eeintech, 2022-08-08:
@KenwoodFox I just migrated to Python3.10 and have no issue running Ki-nTree with this version. Isn't `pip` supported on your machine?

KenwoodFox, 2022-08-09:
Im using python 3.10.5 and im up to date iirc, i try not to install packages except from the aur or my distro's repos, not sure what the issue is.. it does work in a venv, does anybody else using the AUR package have the same issue? maybe its just out of date.

eeintech, 2022-08-15:
Hello @KenwoodFox - I believe `0.5.4` should fix your issue: https://github.com/sparkmicro/Ki-nTree/releases/tag/0.5.4

Unfortunately I do not update the AUR package, I think @randrej does: https://aur.archlinux.org/packages/python-kintree

close2, 2022-08-20:
I've updated to 0.5.4 and still experience this issue.
```
kintree
Traceback (most recent call last):
  File "/usr/bin/kintree", line 33, in <module>
    sys.exit(load_entry_point('kintree==0.5.4', 'console_scripts', 'kintree')())
  File "/usr/bin/kintree", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/usr/lib/python3.10/site-packages/kintree/kintree_gui.py", line 15, in <module>
    from .search import digikey_api as digikey_api
  File "/usr/lib/python3.10/site-packages/kintree/search/digikey_api.py", line 3, in <module>
    import digikey
  File "/usr/lib/python3.10/site-packages/digikey/__init__.py", line 2, in <module>
    from digikey.v2.api import (search, part)
  File "/usr/lib/python3.10/site-packages/digikey/v2/api.py", line 7, in <module>
    from digikey.v2 import models
  File "/usr/lib/python3.10/site-packages/digikey/v2/models.py", line 8, in <module>
    from schematics.exceptions import ConversionError, DataError, ValidationError
  File "/usr/lib/python3.10/site-packages/schematics/__init__.py", line 6, in <module>
    from . import deprecated
  File "/usr/lib/python3.10/site-packages/schematics/deprecated.py", line 8, in <module>
    from .types.serializable import Serializable
  File "/usr/lib/python3.10/site-packages/schematics/types/__init__.py", line 2, in <module>
    from .base import *
  File "/usr/lib/python3.10/site-packages/schematics/types/base.py", line 19, in <module>
    from collections import Iterable, OrderedDict
ImportError: cannot import name 'Iterable' from 'collections' (/usr/lib/python3.10/collections/__init__.py)
```

Modified PKGBUILD (for 0.5.4) on arch.

eeintech, 2022-08-22:
Sorry I have no idea what is happening, indeed I wasn't able to reproduce it on my side, the V2 version of the Digi-Key API is not used and obsolete. So the library loading should fetch the V3 files only...

close2, 2022-08-22:
I had to update the `python-digikey-api` package (a dependency of the `python-kintree` AUR package).
Everything works now.

eeintech, 2022-08-22:
Ha glad to hear it works, `0.5.4` should have automatically updated dependencies but again I'm not familiar with AUR based systems.

closed 2022-08-22 by eeintech

KenwoodFox, 2022-08-30:
I dont see 0.5.4 on the aur Yet but installing via pip got me to 0.5.4 and that works now,