sparkmicro/Ki-nTree #97: [Bug] Crash during invenTree part creation, only halfway created parts

Fivefold, 2022-07-31:
I'm trying to create parts in invenTree from digikey. Unfortunately I get a crash during part creation, which leaves me with parts that are successfully created and have basic info like name, image, description and keywords but are missing any parameters, supplier information, etc.

Here's a log of a part creation: 

```
[MAIN]  Connecting to Inventree server←[0m
←[92m[TREE]     Successfully connected to InvenTree server (ENV=TESTING)←[0m

--------------------
←[95m
[MAIN]  Using Digi-Key cached data for STS4DNFS30L←[0m
←[95m
[MAIN]  Creating part in Inventree←[0m
[INFO]  Category: "Transistors"
[INFO]  Subcategory: "Semiconductors"
[INFO]  User Category: "Active Components"
[INFO]  User Subcategory: "Semiconductors"
←[93m[INFO]     Warning: Parameter map for "Active Components" does not exist or is empty←[0m
←[95m
[MAIN]  Generating Internal Part Number←[0m
[INFO]  Internal Part Number = TRA-000008-00
←[92m[INFO]     Success: Added new part to InvenTree←[0m
Traceback (most recent call last):
  File "c:\program files\python38\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\program files\python38\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\Python38\Scripts\kintree.exe\__main__.py", line 7, in <module>
  File "c:\program files\python38\lib\site-packages\kintree\kintree_gui.py", line 1164, in main
    new_part, part_pk, part_data = inventree_interface.inventree_create(part_info=part_info,
  File "c:\program files\python38\lib\site-packages\kintree\database\inventree_interface.py", line 471, in inventree_create
    category_parameters = inventree_api.get_category_parameters(category_select)
  File "c:\program files\python38\lib\site-packages\kintree\database\inventree_api.py", line 74, in get_category_parameters
    category_templates = category.get_category_parameter_templates(fetch_parent=True)
  File "c:\program files\python38\lib\site-packages\inventree\part.py", line 39, in get_category_parameter_templates
    return self.list(self._api,
  File "c:\program files\python38\lib\site-packages\inventree\base.py", line 139, in list
    if isinstance(response, dict) and response['results'] is not None:
KeyError: 'results'
```

It seems to fail when getting no or the wrong response while fetching parameter templates from invenTree.
Here's the log of the invenTree nginx proxy container: 
https://pastebin.com/9Du8xfHh

I checked the last accessed URL (`GET /api/part/category/6/parameters/?fetch_parent=True`) in my browser and get a full response by the django API.

**Steps to reproduce**
1. install Ki-nTree via pip
2. run the setup (`python3 -m kintree.setup_inventree`)
*Note that I get a few errors here during parameter creation, see below*
3. Configure and verify both invenTree connection and digikey API connection within Ki-nTree
4. Try to add a part

## Version information

Ki-nTree: 0.5.3
InvenTree-Version: 0.7.7

## Errors during setup

```
[MAIN]  Loading Parameters←[0m
ERROR:inventree:POST request failed at 'http://192.168.0.15:5555/api/part/parameter/template/' - 400
ERROR:inventree:{"name":["Illegal character in template name (()"]}
ERROR:inventree:Error creating new object
←[93m[TREE]     Warning: Parameter "(Min) Output Voltage" already exists←[0m
←[92m[TREE]     Success: Parameter "Antenna Type" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "B Constant" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Breakdown Voltage" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Capacitance" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Clamping Voltage" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Collector Gate Voltage" was added to InvenTree←[0m
ERROR:inventree:POST request failed at 'http://192.168.0.15:5555/api/part/parameter/template/' - 400
ERROR:inventree:{"name":["Illegal character in template name (-)"]}
ERROR:inventree:Error creating new object
←[93m[TREE]     Warning: Parameter "Collector-Gate Voltage" already exists←[0m
←[92m[TREE]     Success: Parameter "DC Resistance" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "ESR" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Footprint" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Forward Voltage" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Frequency" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Frequency Stability" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Frequency Tolerance" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Function Type" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Interface Type" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "LED Color" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Load Capacitance" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Locking" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Mating Height" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Max Input Voltage" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Max Output Voltage" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Maximum Gate Voltage" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Memory Size" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Min Input Voltage" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Min Output Voltage" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Mounting Type" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Number of Channels" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Number of Contacts" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Number of Elements" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Number of Rows" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Orientation" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Output Current" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Output Type" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Package Height" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Package Size" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Package Type" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Pitch" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Polarity" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Quiescent Current" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "RDS On Resistance" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "RDS On Voltage" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Rated Current" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Rated Power" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Rated Voltage" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Saturation Current" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Shielding" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Standoff Voltage" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Symbol" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Temperature Grade" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Temperature Range" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Tolerance" was added to InvenTree←[0m
←[92m[TREE]     Success: Parameter "Value" was added to InvenTree←[0m
```

eeintech, 2022-08-01:
Hello there @Fivefold - Thanks for filling in an issue with a lot of details!

So I noticed two things are happening here:
1. You do not have a list of parameters defined for the category "Active Components" in the config file `~/.config/kintree/user/supplier_parameters.yaml` -> This is fine but I would advise to have one, see example here: https://github.com/sparkmicro/Ki-nTree/blob/main/kintree/config/search/supplier_parameters.yaml
Also, this is your short term fix.
3. The `get_category_parameter_templates()` method in InvenTree python API is not returning category parameters anymore... It will need to be fixed and I have opened an issue for it: https://github.com/inventree/inventree-python/issues/131

Fivefold, 2022-08-03:
Can report that it "works" now without any changes on my part after I upgraded to InvenTree 0.8.0 (even though the fix in https://github.com/inventree/inventree-python/issues/131 is supposed to be merged only in 0.8.1)

There are some new POST request-related errors in the log for `/api/company/part/manufacturer/` and `/api/company/part/` but the part _seems_ to be created fully, at least parameters and some supplier-related info are added. Ki-nTree is not crashing anymore Also there are errors on the invenTree side now. Since I don't have 0.8.1 yet and there is a pending update to Ki-nTree as well I will refrain from posting the logs unless you want to see them.

I will report back once 0.8.1 is added as a docker image.

---

> 1. You do not have a list of parameters defined for the category "Active Components" in the config file `~/.config/kintree/user/supplier_parameters.yaml` -> This is fine but I would advise to have [...]

Since I don't see any references to the parameter list in the readme, is this something that's supposed to be created by the setup script or by hand? And where would one find the correct parameter terms to be auto-populated when fetching from a distributor aside from the default supplier_parameters.yaml? What's the intended workflow here?

eeintech, 2022-08-04:
> Since I don't see any references to the parameter list in the readme, is this something that's supposed to be created by the setup script or by hand? And where would one find the correct parameter [...]

1. Yes it has to be created by hand
2. If you add a new category, you will have to populate it like others, eg:
``` yaml
CATEGORY_NAME:
  INVENTREE_PARAMETER_NAME:
    - SUPPLIER_PARAMETER_NAME_1
    - SUPPLIER_PARAMETER_NAME_2
```
Ki-nTree fetches the supplier parameters names and values, then matches them to InvenTree parameters using this file.

For instance, if you search a Digi-Key capacitor, those are the Digi-Key names for parameters: Capacitance, Tolerance, Package / Case, Voltage - Rated, etc.
But then if you go on Mouser or other distributors, the names are different.
So this file unifies the parameter naming across all suppliers/distributors so that, in InvenTree, you can use the parametric table to find the right part and not crumble under duplicate parameters (eg. one capacitor would have "Voltage Rated" for parameter while the others have "Rated Voltage", when they are the same).

The config file I made for this repo are meant to be tweaked and modified according to your own preferences :wink:

eeintech, 2022-08-04:
BTW I found another issue with upgrading InvenTree python API to `0.8.1`: https://github.com/inventree/inventree-python/issues/134

Maybe I'll have time to look into it myself in the coming days.

Fivefold, 2022-08-10:
Thank you for the explanation of the supplier_parameters.yaml! This makes a lot of sense. Maybe it would be a good idea to add this to the README [usage instructions](https://github.com/sparkmicro/Ki-nTree#usage-instructions) since they imply (at least to me) that it's enough to change categories.yaml and run the setup script.

I could make a PR but since the README is the first thing a user sees you probably want to change it as _you_ see fit.

---
Aside from that I tried out InvenTree 0.8.1 https://github.com/inventree/inventree-python/releases/tag/0.8.1 and there is still the same error if it needs to fetch parameters. I see you are already working on it here https://github.com/sparkmicro/Ki-nTree/pull/98. Looking forward to when it's done, thanks for your work!

eeintech, 2022-08-22:
I forgot I wanted to update the README, will keep this issue opened in the meantime as a reminder.

PR #98 (2022-08-15): Update InvenTree API to fix parameters fetching
Fixes #97, fixes #94
commit 3ca758dd (2022-08-15): Fix #97
PR #101 (2022-11-11): 0.6.0
## MAJOR RELEASE

### New features
- Supplier/manufacturer alternates
  - On main GUI window, added an option for adding supplier and/or manufacturer alternate to an InvenTree part
  - When this option is enabled, the part creation flow is simplified to adding the supplier/manufacturer data to an already exisiting InvenTree part using its ID or IPN field
  - This option only works with InvenTree parts as the concept of supplier/manufacturer alternates does not exists (yet) in KiCad
- Manufacturer name match
  - Some suppliers use slightly different names for the same manufacturer
  - Add [... cut]
closed 2022-11-11 by eeintech