sparkmicro/Ki-nTree #99: FR Price breaks

Weissnix4711, 2022-08-15:
Would it not be useful to import pricing information from mouser, lcsc and digikey into InvenTree?

I do know InvenTree's pricing system is currently being overhauled, so this would possibly be a FR for when 0.9 releases.

eeintech, 2022-08-15:
@Weissnix4711 IMO pricing should be integrated as part of InvenTree supplier plugins, the price of parts is constantly moving and Ki-nTree is only ran at the creation of the part, the price would become obsolete very quickly.

Weissnix4711, 2022-08-19:
That makes sense. Are you aware of any current effort to create supplier plugins for LCSC specifically? Or other suppliers, but I'm mostly interested in LCSC.

Or should we wait for the pricing system overhaul (and major changes to the plugin system and API which would likely follow) first?

eeintech, 2022-08-20:
No I'm not aware of any support for LCSC. I am not sure when the pricing system overhaul will be completed, but the plugin system is already in place.

closed 2022-08-22 by eeintech