sparkmicro/Ki-nTree #106: LCSC  Warning: API data download failed

PricelessToolkit, 2022-12-03:
Hi, First of all, I want to say a huge thank you for this great tool!
I think after today's LCSC update "shop in maintenance mode 5+ hours"  Ki-nTree cant fetch data. 
I have tried several parts numbers C2935912, C383211, C2907011 .

the terminal gives the following errors

[MAIN]  LCSC search for C2907011
[INFO]  Warning: API data download failed (HTTP Error)
[INFO]  Error: Failed to fetch data for "C2907011"

Maybe they changed something?

PricelessToolkit, 2022-12-05:
ok, Thanks!

PricelessToolkit, 2022-12-07:
> Related I think: https://blog.honzamrazek.cz/2022/12/jlcparts-falling-down-to-ashes/
> 
> So it seems like JLC PCB and LCSC discontinued access to their API... I have sent a message to LCSC via their website, let's see if I get a response...

Yesterday I sent a message too

eeintech, 2022-12-08:
I am not really sure either why cutting legs on everybody would be a good idea, without instating a new option.

I have responded to their message above and they said: "We will provide API access to our long-term cooperation customers" and then asked for my account email address to keep me posted.
Not sure where this is going to lead...

PricelessToolkit, 2022-12-09:
if it's not difficult, let me know if there is news. Thanks

closed 2022-12-12 by eeintech

PricelessToolkit, 2022-12-24:
After changing the URL everything works again.
Thanks

eeintech, 2022-12-27:
@PricelessToolkit Check out the [`0.6.3` release](https://github.com/sparkmicro/Ki-nTree/releases/tag/0.6.3)

PR #107 (2022-12-12): 0.6.2
- Disabled LCSC in list of suppliers: fixes #106 
- Updated SnapEDA data fetch to use download method from tools (so much faster!)