sparkmicro/Ki-nTree #109: Error while installing

Mimosa-oceanis, 2022-12-20:
Hi, I'm having trouble installing Ki-nTree. 
I run `pip install -U kintree` in my terminal (I'm using Mac OS)
and this is what I got in return: 

```
ERROR: Ignored the following versions that require a different python version: 0.4.0b0 Requires-Python >=3.7,<3.10; 0.4.0rc0 Requires-Python >=3.7,<3.10; 0.4.0rc1 Requires-Python >=3.7,<3.10; 0.4.1 Requires-Python >=3.7,<3.10; 0.4.2 Requires-Python >=3.7,<3.10; 0.4.3 Requires-Python >=3.7,<3.10; 0.4.4 Requires-Python >=3.7,<3.10; 0.4.5 Requires-Python >=3.7,<3.10; 0.4.6 Requires-Python >=3.7,<3.10; 0.4.7 Requires-Python >=3.7,<3.10; 0.4.8 Requires-Python >=3.7,<3.10; 0.5.0 Requires-Python >=3.7,<3.11; 0.5.1 Requires-Python >=3.7,<3.11; 0.5.2 Requires-Python >=3.7,<3.11; 0.5.3 Requires-Python >=3.8,<3.11; 0.5.4 Requires-Python >=3.8,<3.11; 0.6.0 Requires-Python >=3.9,<3.11; 0.6.1 Requires-Python >=3.8,<3.11; 0.6.2 Requires-Python >=3.8,<3.11
ERROR: Could not find a version that satisfies the requirement kintree (from versions: none)
ERROR: No matching distribution found for kintree
```

I'm running python Python 3.10.8

eeintech, 2022-12-20:
Hello @Mimosa-oceanis Could you please post the full traceback?
Also can you please post the result of this command: `pip --version`

Mimosa-oceanis, 2022-12-21:
This is all what I get in the terminal. 

and my pip version is: 
`pip 22.3.1 from /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pip (python 3.11)`

eeintech, 2022-12-21:
Your pip version indicates it uses `Python 3.11` which is not included in in `>=3.8,<3.11` required Python versions.

You can try to run the pip executable that came with your `Python 3.10` instead, could it be at `/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pip install kintree` works?

eeintech, 2022-12-27:
@Mimosa-oceanis Any update on your side?

eeintech, 2022-12-30:
@Mimosa-oceanis Sounds good, I'm sure you'll figure it out!

closed 2022-12-30 by eeintech