sparkmicro/Ki-nTree #112: Error on startup after clean install

Fotte, 2023-01-04:
Hey guys.

I just wanted to try Ki-nTree out and fired up a fresh xubuntu 20.04.1 VM. My problem is I can not start kintree.

kintree Version: 0.6.4
Python Version: 3.10.6
Installed via pip using virtual environment.

Error on startup:
`
(env-kintree) f@f-virtual-machine:~$ kintree
[ERROR] some Ki-nTree configuration files seem to be missing
`

I did not change the config files (nor did I copy a basic set). Am I supposed to do that before the first startup? I could not find it in the documentation :(

eeintech, 2023-01-05:
Hello @Fotte 

This is my bad! I introduced a bug in the latest release (`0.6.4`) which I will fix in `0.6.5`, I'll keep you posted when released!

closed 2023-01-05 by eeintech

eeintech, 2023-01-05:
`0.6.5` is now on PyPI, let me know if it works on your machine now :+1:

Fotte, 2023-01-05:
Hi @eeintech
Thanks for your reply. The updated version works. I still had to manually install python3-tk - this is probably obvious for python people?

Anyway: kintree starts up now. Thank you for your effort!

eeintech, 2023-01-06:
I'm gonna take a guess here as I don't know xubuntu but I imagine Python 3.10 is not their default/supported version and you have probably installed it yourself, which in this case was probably the reason why you were missing Tkinter packages. In supported versions this shouldn't be needed, which is the case for most people.

PR #114 (2023-01-05): 0.6.5
- Fixes #112 (regression bug, introduced in `0.6.4`)
- Fixes #113 and a "New InvenTree Part" detection bug
- Fixes crash when no Farnell/Newark/Element14 store is selected and display a blank selection instead (ref: [this thread](https://forum.kicad.info/t/ki-ntree-fast-part-creation-in-kicad-and-inventree/24300/40))
- Fixed error message when Farnell/Newark/Element14 supplier search fails:

[image]

- Added overwrite of manufacturer name with matching database entry (partial ratio) per feedback from Kedarius in [this thread](https://forum.kicad.info/t/ki-ntree-fast-part-creation-in-kica [... cut]