sparkmicro/Ki-nTree #113: Part already in Inventree database, different MPN

martin323232, 2023-01-05:
For some components with different MPN and even some with different manufactuerer I get "part already in Inventree database"

GR67M20A and GR67M25A gets added to the same part in inventree.

[image]

martin323232, 2023-01-05:
Had to add a different name for the two grommets.

eeintech, 2023-01-05:
Hi there @martin323232 

Here the problems is that those two parts can be differentiated as their name and description are the same on Digi-Key... and InvenTree does not accept a new part when those are the same.
The only way you can create a new part is to change the name, for instance for `GR67M20A`, you can append the MPN at the end of the name:
[image]

Before you do that, I just found a bug with that that I am going to fix in `0.6.5`, give me a bit of time to release it, I'll let you know :smiley: 

> even some with different manufactuerer

If an InvenTree part already contains a manufacturer part with the same manufacturer **and** MPN then it will be skipped and won't be added to the database. In other cases:
- If manufacturer name is different: it will always create a **new manufacturer part** but decision to create a new InvenTree part depends if its "name" and "revision" field are the same to other existing InvenTree part
- If MPN is different: it will **always** create a new InvenTree part

closed 2023-01-05 by eeintech

eeintech, 2023-01-05:
`0.6.5` is now on PyPI, let me know if the workaround works for you :+1:

PR #114 (2023-01-05): 0.6.5
- Fixes #112 (regression bug, introduced in `0.6.4`)
- Fixes #113 and a "New InvenTree Part" detection bug
- Fixes crash when no Farnell/Newark/Element14 store is selected and display a blank selection instead (ref: [this thread](https://forum.kicad.info/t/ki-ntree-fast-part-creation-in-kicad-and-inventree/24300/40))
- Fixed error message when Farnell/Newark/Element14 supplier search fails:

[image]

- Added overwrite of manufacturer name with matching database entry (partial ratio) per feedback from Kedarius in [this thread](https://forum.kicad.info/t/ki-ntree-fast-part-creation-in-kica [... cut]