sparkmicro/Ki-nTree #116: Quick way to open the part page on the supplier site in the system browser

Kedarius, 2023-01-12:
It would be great if one could open the part page on the supplier website with a single click from the part dialog.

For example I've been adding some 74 logic gates and the description from the API was something like "74xyz logic TTL gate....". As I am not so versed in all the 74 numbers, I prefer to have something like "hex inverter" in the name. But for that I need to open the browser window and search for the part. It would be nice if the part dialog had a button or something to open the URL in the browser.

eeintech, 2023-01-13:
Hello @Kedarius 

If you are referring to Farnell/Newark/Element14 parts, their API does not provide the part webpage URL in the response... I asked them about it and they confirmed it isn't accessible.

If you are using any other supplier, you can get the URL in the part form:
[image]

Copy-paste to browser and you're good to go!

Kedarius, 2023-01-14:
Thanks for your answer. The point of my suggestion was to speed up the process, sure I can copy/paste into browser, but having an one click button is much faster. 
And it seems that at farnell you can actually go to https://cz.farnell.com/1003248 and it will redirect you to the correct page. But it is not in the official API and probably is not guaranteed. However it would be better to have the possibility to have the farnell links in this form instead of just URL of the whole website.

eeintech, 2023-01-16:
Hi @Kedarius For the button I tend to not be onboard on adding excessive feature that solve little time, most people will look for part on the supplier website **then** add it to their database so it is rare you need to access the webpage of the part you're creating as you should have it open(ed) already.

For the supplier link with Farnell&co, I did ask their API support team if there as a quick access URL available and they gave me a negative answer, therefore I just put the base store URL.
It does look like appending the part number redirects to `xyx.com/w/search?st=<mpn>` so this could potentially work :+1: 
I can cut a new release for this, keep you posted!

Kedarius, 2023-01-17:
Ok, it is not a big deal, I just thought that small button near the URL input field would be handy. 

Just for your info, how I use ki-nTree (and inventree). We do mainly SW development so HW is not our main priority and therefore we tend to outsource most of the production (typically about 10-100 units) to assembly house. But for R&D we need to keep some sort of stock overview. Most of the parts will never make it to be used in production and the part management should be simple and efficient and not prohibitive.  So for example I do not keep track of parts quantity, just where it should be stored.... And only if the part make it to the production then I keep it in separate storage box and I keep track of the quantity or have the part serialized. The same is with datasheets, ECAD models, etc....
So basically I need to have a quick way to enter the part in the system. Now it takes me about 2-3 min per part so I am looking for ways where to make it faster. I start with search in inventree using SKU, then if it is not found then I go directly to ki-ntree and I add it to inventree. In ki-ntree I edit the name and description to be as descriptive as it can be on the printed label later. Then I print the label and set 1 unit in stock at correct storage bin. So most of the time I do not even go to the supplier page. If the part gets used for some experimenting/production then I gradually improve the inventree entry (add more attachments, notes) and I do create/find KiCad footprint and symbol. And  those I do keep in git repo....
Sorry for longer post I just wanted to tell you how I use your software....

closed 2023-01-17 by Kedarius

eeintech, 2023-01-17:
@Kedarius Well this is really interesting feedback! Thanks for taking the time to share, I have shared your comment with the InvenTree maintainers too as they want to know how InvenTree users use their tool.

PR #117 (2023-02-16): 0.6.6
- Fixes #116: corrected supplier URL for Element14 companies
- Fixes empty KiCad symbol name when disabling IPN auto-generation
- Template fix
- Dependencies update

:warning: Last release to support KiCad V6