sparkmicro/Ki-nTree #118: Kicad 6-7

mikhailmihalkov, 2023-01-19:
Can I add support for SamacSys? ( https://componentsearchengine.com/ )

eeintech, 2023-01-19:
Hello @mikhailmihalkov Sure, do they have an API?

mikhailmihalkov, 2023-01-19:
I can't say for sure but there is such a plugin https://github.com/olback/library-loader There is also a bootloader application on the company's website

mikhailmihalkov, 2023-01-20:
eeintech please clarify what questions I need to ask for development I will try to send a request

mikhailmihalkov, 2023-01-20:
eeintech Another question) is it possible to load 3d component models through this script?

eeintech, 2023-02-16:
> eeintech Another question) is it possible to load 3d component models through this script?

Anything is possible whenever an API is available :smiley: 

> eeintech please clarify what questions I need to ask for development I will try to send a request

You need to ask them if they have a public API available for use:
- do they accept user registration
- do they provide part search endpoint
- etc.

eeintech, 2023-02-22:
@mikhailmihalkov I just contacted them using their contact form: https://componentsearchengine.com/contact-us
If they don't respond in few days, I'll try their email address: info@SamacSys.com

eeintech, 2023-03-06:
FYI- Email sent, with little hope to get a response :confused:

mikhailmihalkov, 2023-03-19:
I have access to their plugins for different programs look maybe there is a hint there....
[LibraryLoaderSetup2v50.msi.zip](https://github.com/sparkmicro/Ki-nTree/files/11010393/LibraryLoaderSetup2v50.msi.zip)

[AltiumLL2v20.zip](https://github.com/sparkmicro/Ki-nTree/files/11010394/AltiumLL2v20.zip)
[KiCad_Plugin1v40.zip](https://github.com/sparkmicro/Ki-nTree/files/11010395/KiCad_Plugin1v40.zip)
[EAGLE_Plugin2v20.zip](https://github.com/sparkmicro/Ki-nTree/files/11010396/EAGLE_Plugin2v20.zip)

eeintech, 2023-03-29:
After multiple attempts for contact, there is no response so I will close this ticket now.

@mikhailmihalkov If you ever get more info, feel free to reach out and re-open this issue.

closed 2023-03-29 by eeintech