sparkmicro/Ki-nTree #132: Hello there,

crossband, 2023-03-02:


eeintech, 2023-03-02:
Hello @crossband 

I was able to reproduce it, and here's the process I followed:
1. Navigate to https://visualstudio.microsoft.com/visual-cpp-build-tools/
2. Click on "Download Build Tools" button
3. In the installer, in the "Visual Studio Build Tools 2022" card, click "Modify"
[image]
4. Click on "Individual Components" tab then install the following components:
[image]
5. Re-install Ki-nTree
```shell
pip install - U kintree
```

It worked for me. However I know this is not ideal installation flow, I hate to see how complicated this single library makes things for you and others so I'm going to attempt to remove it for `1.0.0` version.
Thanks for bringing this to my attention, let me know if you have any other question.

crossband, 2023-03-02:
Thanks for the quick feedback. Following the instruction I was able to install it.

eeintech, 2023-03-02:
Awesome to hear, sorry again for the trouble!

closed 2023-03-02 by eeintech