sparkmicro/Ki-nTree #135: Issue Adding Parts

arkajyotidas, 2023-03-05:
Either its Resistors or Capacitors, Whenever I am trying to add parts I am getting this error

Error loading template file (C:/Program Files/KiCad/6.0/share/kicad/template/capacitor.kicad_sym)←[0m

Error loading template file (C:/Program Files/KiCad/6.0/share/kicad/template/resistor-sm.kicad_sym)←[0m

eeintech, 2023-03-06:
Hello @arkajyotidas 

Are the template files located in this folder? eg. `C:/Program Files/KiCad/6.0/share/kicad/template/`
Can you open one the file and see if it has the same content as in the provided file in this repo?
For instance: https://github.com/sparkmicro/Ki-nTree/blob/main/kintree/kicad/templates/capacitor.kicad_sym

arkajyotidas, 2023-03-06:
so I need to place these templates in my kicad folders.

eeintech, 2023-03-06:
Yes if the file is not found or empty then you get the error.

arkajyotidas, 2023-03-06:
When's KiCad 7 support coming?

eeintech, 2023-03-06:
Did you fix this issue? Can I close now?

For KiCad V7, it will come when this milestone is complete: https://github.com/mvnmgrx/kiutils/milestone/2

closed 2023-03-06 by arkajyotidas