sparkmicro/Ki-nTree #136: TypeError: 'type' object is not subscriptable after upgrade to 1.0.0b1

Kedarius, 2023-03-07:
I've upgraded to 1.0.0b1 with
`pip install -U kintree --pre`
and now kintree does not start:
```
Traceback (most recent call last):
  File "c:\program files\python38\lib\runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\program files\python38\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\Python38\Scripts\kintree.exe\__main__.py", line 4, in <module>
  File "c:\program files\python38\lib\site-packages\kintree\kintree_gui.py", line 3, in <module>
    from .gui.gui import kintree_gui
  File "c:\program files\python38\lib\site-packages\kintree\gui\gui.py", line 3, in <module>
    from .views.common import update_theme, handle_transition
  File "c:\program files\python38\lib\site-packages\kintree\gui\views\common.py", line 178, in <module>
    class SwitchWithRefs(ft.Switch):
  File "c:\program files\python38\lib\site-packages\kintree\gui\views\common.py", line 185, in SwitchWithRefs
    refs: list[ft.Ref] = None,
TypeError: 'type' object is not subscriptable
```
I've tried to remove my configuration but it did not help.

eeintech, 2023-03-07:
Hello @Kedarius 

I haven't been able to reproduce your issue... I am not sure what is the problem :confused: 

Do you mind to comment that line 185 in the `c:\program files\python38\lib\site-packages\kintree\gui\views\common.py` file and re-try?

Kedarius, 2023-03-07:
Hi @eeintech ,
commenting out did not help as it said
`NameError: name 'refs' is not defined`
however setting it to [] helped (I also needed to do the same on line 215). See the attached file with modifications.
[common.zip](https://github.com/sparkmicro/Ki-nTree/files/10914310/common.zip)

eeintech, 2023-03-08:
@Kedarius Thanks, I will try to understand the underlying of this issue and fix it, thanks for the report.

eeintech, 2023-03-10:
@Kedarius I found the issue (this type hinting was not supported in Python=3.8) and added a GUI test to catch it next time :smile:
It will be fixed with `1.0.0b2`!

commit 9c2856ff (2023-03-10): Now fix #136
PR #138 (2023-03-23): 1.0.0b2
- Fixes #136
- Fixes #137 
- Fixed bug creating custom part with IPN setting enabled
- Added GUI CI test to catch errors launching the GUI
- Replaced `wrapt_timeout_decorator` package with version from PyPI
- Updated dependencies
closed 2023-03-23 by eeintech