sparkmicro/Ki-nTree #139: 1.0.0b1 category does not exist

martin323232, 2023-03-15:
Upgraded to beta version because my usual way of adding new categories stopped working. 
After adding a new category to digikey_categories.yaml it overwritten by the program. 

Have had this issue before, but can’t remember how I fixed it. 

In beta I download all categories and after selecting the correct one it says it does not exist, have tried different categories with same result. 
Tried to take an existing part and adding and this gives "category does not exist"

[image]

eeintech, 2023-03-15:
Hello @martin323232 

I tried to reproduce your issue but I'm sorry I can't...

1. I added the categories to my setup:

[image]

2. In InvenTree view, I clicked on "Reload InvenTree Categories" button

[image]

3. Categories were displayed in the dropdown, I selected Thermal Pads (also added a new category code but that has nothing to do with categories)

[image]

4. I launched the part creation in the Create view and it ran fine

[image]

Am I missing some step that you did on your end?
Can you please make sure your Ki-nTree version is `1.0.0b1` and InvenTree is `0.9.x` or later?

Thanks!

martin323232, 2023-03-16:
It was an issue with my Inventree version, updated to 0.10.1 and got it working.
Thanks for a great addon program to Inventree and KiCad.

eeintech, 2023-03-17:
Great to hear! Thanks @martin323232 

Closing now :smiley:

closed 2023-03-17 by eeintech