sparkmicro/Ki-nTree #140: [feature-request] support proxies for inventree access

T0jan, 2023-03-15:
Hi,

since the last [version](https://github.com/inventree/inventree-python/releases/tag/0.10.1) the inventree python api supports the definition of proxy addresses to for example simplify the access to the server from a different network. It would be nice if Ki-nTree also could support this e.g. with an additional input field in the inventree config.

eeintech, 2023-03-15:
Hello @T0jan 

To clarify your request, what exactly would you need to be added to the GUI (Text field?) and InvenTree API (replace Server URL?) in the current code? I'm not sure how the proxy works so please share as much information as you can, also I am not sure how to test this on my end... :confused:

T0jan, 2023-03-16:
@eeintech sure, no problem. I have to show you on the old gui thos, as the v1.0.0 preview does not work on my side right now.

so to add proxies I would add another text field in this dialog:

[image]

where the user either can enter his proxy servers or leave it empty. A typical input when using proxies would be 

```http=socks5h://localhost:9000, https=socks5h://localhost:9000```

this then needs to be converted to a dictionary in the form:

```proxies = {'http':'socks5h://localhost:9000', 'https':'socks5h://localhost:9000'}```

for the python request module which handles all the requests of the InvenTreeAPI class.

The information of which proxy to use is set during class initialization and every request afterwards uses the class attribute containing the dictionary. so you only have to add `proxies=proxies` to 

https://github.com/sparkmicro/Ki-nTree/blob/bb05c74444a19bf839d6cbaab4d322b8a544f2ab/kintree/database/inventree_api.py#L29

and obviously add another parameter `proxies` to the connect function.

If you don't want to create your own proxy server or use an existing one for your testing I also can help you with testing if you want.

While you already work on the InvenTree settings dialog it would be also nice if one could use the token verification method instead of username and password. e.g. if the username field is left empty the password will be used as token instead or something like this.

eeintech, 2023-03-17:
Thanks @T0jan I appreciate you taking the time to explain in details as I'm not familiar with the matter.

Couple of options:
- Do you think you can work on it and squeeze a PR before I go ahead and release `1.0.0` in the coming weeks?
- We assign it to `1.1.0` and I take a stab at it after `1.0.0` release

I'm sorry I'm quite busy right now and I don't want to delay too much the release of `1.0.0` for new features at the moment.

T0jan, 2023-03-21:
@eeintech I am not certain when I can work on it so I definitely wouldn't delay `v1.0.0` for this feature. if I make it in time it would be nice but as I have not that much experience in GUI programming it could take me a while. So no worries, nothing on my side depends on having this asap included. :)

closed 2023-04-24 by eeintech

eeintech, 2023-04-24:
@T0jan Released in `1.0.1` :wink:

PR #153 (2023-04-24): 1.0.1
 - Added extra fields support for all suppliers
 - Fixes #140 (after #149 is merged)