sparkmicro/Ki-nTree #144: Using custom name as component name

T0jan, 2023-04-03:
hi again,

while working on the feature request a question popped up in my head: is it possible to have a custom name for the kicad symbol /inventree entry?

What I want to achieve: in the kicad library a resistor should be named e.g. 10k@1%

I tried a bit with modifying the templates but not really got anywhere so before wasting more time I wanted to ask if it is even possible right now to achieve this without sacrificing the MPN field to set the name?

eeintech, 2023-04-04:
Hello @T0jan 

Right, I never thought of it before so there is no workaround at the moment. 

I have to ask: how do you imagine it would work?

Thinking of it quickly, I say it is pretty equivalent to manually add the symbol name in Ki-nTree vs edit it in KiCad, the only advantage of doing it in Ki-nTree is that you wouldn't forget to do it in KiCad.

T0jan, 2023-04-05:
@eeintech exactly I fear of forgetting it, then adding it to the schematics without thinking and after that fixing the name would be a pain with how kicad handles name changes of parts.

My idea was either to add a simple switch in the settings where you can change between MPN field as name and the name field. a big more extended version would be instead of this a drop down where the user can select which field to use as name.

T0jan, 2023-10-23:
was added with #172, closing here.

closed 2023-10-23 by T0jan