sparkmicro/Ki-nTree #150: Categories not populated

Scowt, 2023-04-19:
Hello,

As far as i can tell, I've run the setup scripts correctly. I can access inventree via localhost, and ki-ntree can connect to it.
But although the script reports that various parameters already exist, it does not appear to create any categories.

`python3 -m kintree.setup_inventree`

```
[MAIN]	Starting InvenTree setup
[MAIN]	Connecting to Inventree
[TREE]	Successfully connected to InvenTree server (ENV=DEVELOPMENT)

[MAIN]	Loading Parameters
[TREE]	Warning: Parameter "Antenna Type" already exists
[TREE]	Warning: Parameter "B Constant" already exists
<snip>
[TREE]	Warning: Parameter "Value" already exists
```

eeintech, 2023-04-19:
Hello @Scowt 

Weird, it works fine for me.

What's in your `~/.config/kintree/user/categories.yaml` file?

Scowt, 2023-04-19:
Thanks for the quick reply!

$ cat ~/.config/kintree/user/categories.yaml
CATEGORIES: {}
CODES:
  Assemblies: PCA
  Capacitors: CAP
  Circuit Protections: PRO
  Connectors: CON
  Crystals and Oscillators: CLK
  Diodes: DIO
  Inductors: IND
  Integrated Circuits: ICS
  Mechanicals: MEC
  Miscellaneous: MIS
  Modules: MOD
  Power Management: PWR
  Printed-Circuit Boards: PCB
  RF: RFC
  Resistors: RES
  Transistors: TRA

Scowt, 2023-04-20:
Problem solved.

I tried using the inventree user instead of my regular username, ran kintree to set up the inventree server, then kintree_setup_inventree.

Successfully added all categories after that, all are visible to my regular username also.

eeintech, 2023-04-20:
Good to hear! The categories dictionary was indeed empty :confused:

closed 2023-04-20 by eeintech