sparkmicro/Ki-nTree #151: Feature request: Set initial stock level when part created

Scowt, 2023-04-21:
We have an existing (disorganised...) lab, with many components in boxes. 

Adding all these will be time consuming no matter what; something that could speed up the workflow would be an option to create a stock item when adding a new part via Ki-ntree.

Perhaps something like a quantity text field; if value entered, create stock item & set stock level.

eeintech, 2023-04-21:
Should be straightforward and can be added in coming dot release

Scowt, 2023-04-21:
I'd be happy to have a crack at it, sure

martonmiklos, 2023-09-01:
FYA I implemented this in this branch:

https://github.com/martonmiklos/Ki-nTree/tree/add_option_to_create_stock_items

I will submit a PR after some polishing.
Feedback on the UI/implementation is welcome.

PR #215 (2024-03-05): 1.0.7
* #212 
* #216 (fixes #151)
* Fix #209
closed 2024-03-05 by eeintech