sparkmicro/Ki-nTree #152: Feature request: allow part search to be less exact

Scowt, 2023-04-21:
Regarding part search using Digikey (I have not attempted other suppliers yet) - would it be possible to search on a smaller fragment of the part number? 

It seems Ki-nTree needs the search to return only one result to be successful, which I totally get from a programming standpoint but is a mild pain point when bulk-adding existing stocked parts.

To clarify, I am being lazy and trying to cut down on the data entry task; typing AUIRFR540ZTRL results in one component, but typing AUIRFR5 takes half as long and would only return 13 components, which is pretty easy to select between. 

I could imagine that as further characters are added, more parts are filtered out from the one initial search, to reduce api calls.

This is definitely not a huge priority, more of a UX comment than anything else. If there is an alternative method I'd be happy to hear it.

Thank you for all your hard work!

eeintech, 2023-04-21:
Hello @Scowt 

I have to experiment with that and can't promise it can be easily added, however I'm thinking a pop-up to select the right part _could_ work, as long as the software creates only one part at the time as it wasn't intended for creating multiple at once.

eeintech, 2024-02-27:
Hello @Scowt Is that still a wish on your end? I didn't get sense this is really a necessity.

Scowt, 2024-02-28:
Hi @eeintech  - place I work for moved away from Kicad / Ki-nTree, so no longer a wish for me. I do think it would add to usability in general, but as nobody else chimed in it's probably not a priority for them either :-)

Thanks for following up

eeintech, 2024-02-28:
Ok thanks for the update @Scowt, closing this issue 😃

closed 2024-02-28 by eeintech