sparkmicro/Ki-nTree #157: LCSC Part Saving Fails

ScaredyCat, 2023-05-19:
I'm working through some of my parts and some of them keep failing, some are perfectly fine.

All my parts are currently coming from LCSC. It seems that the adding to InvenTree fails with 

`[TREE]	Successfully connected to InvenTree server (ENV=DEVELOPMENT)
Exception in thread Thread-471:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/andy/kin/env-kintree/lib/python3.8/site-packages/kintree/gui/views/main.py", line 1196, in create_part
    new_part, part_pk, part_info = inventree_interface.inventree_create(
  File "/home/andy/kin/env-kintree/lib/python3.8/site-packages/kintree/database/inventree_interface.py", line 494, in inventree_create
    inventree_part = translate_form_to_inventree(
  File "/home/andy/kin/env-kintree/lib/python3.8/site-packages/kintree/database/inventree_interface.py", line 285, in translate_form_to_inventree
    parameter_value = part_tools.clean_parameter_value(
  File "/home/andy/kin/env-kintree/lib/python3.8/site-packages/kintree/common/part_tools.py", line 138, in clean_parameter_value
    second_value = split_space[2]
IndexError: list index out of range
`

Example of content returned from the part lookup, eg https://wmsc.lcsc.com/wmsc/product/detail?productCode=C86781  that causes problems. I these some extra try/catch method that could help fix this?

ScaredyCat, 2023-05-20:
I'm wondering is it's to do with non-ascii characters in the output, but don't let this mislead anyone, it's a guess.

eeintech, 2023-06-04:
Hello @ScaredyCat Is it happening for all LCSC parts or only this one example?

ScaredyCat, 2023-06-04:
There are other examples of it. A lot work fine some just fail with that error.

Here are some examples of failing part numbers:

C2843785
C134077 
C51118

The initial data is fetched (pic1) but the import fails (pic2)

[image]

[image]

commit d22dcd8f (2023-08-04): Fix #157
PR #169 (2023-08-04): 1.0.2
- Add #167
  - uploading datasheets to InvenTree, see #162
  - add an update functionality to the alternate mode to add missing parameters, datasheets and images to an existing part
  - add the option to use the pricing data of each supplier to create/update price breaks for a supplier part
- Fix #156 
- Fix #157 
- Fix instance compare Python syntax
- Add README warning about InvenTree `0.12` parameter compatibility issue
- Update PyYAML dependency as older version was generating an error with `poetry` package
closed 2023-08-04 by eeintech